Klipper3d/klipper #845: Implementation of M118 for custom Action Commands

ndupont-net, 2018-11-02:
Hi, as I like to add further Octoprint control via the LCD and trigger some system commands from my gcode, I have implemented M118 as a revised version of ECHO.

It would be great to see that functionnality implemented in Klipper :)

Here's the snippet added to **gcode.py**
`
cmd_M118_when_not_ready = True 
    def cmd_M118(self, params): 
        self.respond_info(params['#original'].replace("M118 ","").replace("//","",1))
`

As self.respond_info prefixed the text with "// ", I discard the first "//" of the argument to allow for a standard M118 //action: 

It's also declared in the handlers section.
` all_handlers = [
        'G1', 'G4', 'G28', 'M18', 'M400',
        'G20', 'M82', 'M83', 'G90', 'G91', 'G92', 'M114', 'M220', 'M221',
        'SET_GCODE_OFFSET', 'M206',
        'M105', 'M104', 'M109', 'M140', 'M190', 'M106', 'M107',
        'M112', 'M115', 'IGNORE', 'GET_POSITION',
        'RESTART', 'FIRMWARE_RESTART', 'ECHO', 'STATUS', 'HELP', 'M118']`

Here are the action command set in Octoprint
[image]

Here is the customized menu that give more control on Octoprint from the LCD (shutdown being the most important)
[image]

KevinOConnor, 2018-11-07:
To contribute code to Klipper you would need to create a github pull request - see: https://github.com/KevinOConnor/klipper/blob/master/docs/CONTRIBUTING.md

I'm not really sure why the M118 is needed though - the menu items listed above don't seem to invoke that command.  In general, we prefer new additions to be contained in a new "extras" module - there is some details on doing this at: https://github.com/KevinOConnor/klipper/blob/master/docs/Code_Overview.md

-Kevin

ndupont-net, 2018-11-08:
Commit tested and validated on my printer before pull request.

I close the issue

closed 2018-11-08 by ndupont-net