Klipper3d/klipper #847: Implementation of M300 (beeper support) that could be included in example-extras.cfg

ndupont-net, 2018-11-02:
Hi, here is how I have implemented the beeper using an output_pin and a gcode macro.

This could be added to **example-extras.cfg** as it would make for a great drag and drop snippet in custom configurations.

M300 [P<ms>] [S<Hz>],  although the frequency is not pitch-perfect, based on 1kHz and cannot be higher than that with this pwm cycle_time.

```
[output_pin BEEPER_pin]
pin: ar37
pwm: True
value: 0
shutdown_value: 0
cycle_time: 0.001
scale: 1000

[gcode_macro M300]
default_parameter_S=1000
default_parameter_P=100
gcode:  SET_PIN PIN=BEEPER_pin VALUE={S}
        G4 P{P}
        SET_PIN PIN=BEEPER_pin VALUE=0
```

My apologises if this is not the desired way to propose new functionnalities.

I also have a Tevo Tarantula config file that could be shared in the examples.

KevinOConnor, 2018-11-07:
FYI, in issue #787 there was a discussion on possibly adding a "config/sample-macros.cfg" file for things like the above.

-Kevin

ndupont-net, 2018-11-07:
That sounds like a good place for it, issue closed then

Nicolas

closed 2018-11-07 by ndupont-net