Klipper3d/klipper #850: Z_TILT_ADJUST does nothing in VORON 1.5

dgcaste, 2018-11-03:
After I home, I run a Z_TILT_ADJUST and it moves to the first probing point and then stops. Klippy.log shows nothing after the command is read. Attached log. Here is the CFG:

`
# This is a Voron V1.5 specific cfg file  with pressure advance modified on 180718 by Michael Lanfranconi
# This file serves as documentation for config parameters of corexy
# style printers. One may copy and edit this file to configure a new
# corexy printer. Only parameters unique to corexy printers are
# described here - see the "example.cfg" file for description of
# common config parameters.

# DO NOT COPY THIS FILE WITHOUT CAREFULLY READING AND UPDATING IT
# FIRST. Incorrectly configured parameters may cause damage.

# this is required for gantry leveling and replaces your G28 command
# with the gcode used here.  Used to home X/Y/Z with mechanical switches
[homing_override]
set_position_z: 0
gcode:
    G90
    G0 Z15 F600
    G28 X0 Y0
    G0 X121 Y117 F3000
    G28 Z
    G0 Z15 F600

# The stepper_x section is used to describe the X axis as well as the
# stepper controlling the X+Y movement.
[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: 0.0125
endstop_pin: ^ar3
position_endstop: 242
position_max: 242
homing_speed: 50
homing_positive_dir: true
#max for VORON X is X242

# The stepper_y section is used to describe the Y axis as well as the
# stepper controlling the X-Y movement.
[stepper_y]
step_pin: ar60
dir_pin: ar61
enable_pin: !ar56
step_distance: 0.0125
endstop_pin: ^ar14
position_endstop: 234
position_max: 234
homing_speed: 50
homing_positive_dir: true
#max for VORON X is Y234

 
[stepper_z] 
# ZR is connected to Z
step_pin: ar46
dir_pin: !ar48
enable_pin: !ar62
step_distance: 0.0025
endstop_pin: ^ar18
position_endstop: -0.77
position_max: 238
#max for VORON X is Z238
position_min: -1
homing_speed: 12

[stepper_z1]
# ZL is connected to E1
# E1 stepper pins are:
[... 195 lines cut ...]

[mcu]
serial: /dev/ttyACM0
pin_map: arduino
#restart_method: rpi_usb
# I don't think my Arduino is USB powered.

[printer]
kinematics: corexy
#   This option must be "corexy" for corexy printers.
max_velocity: 300
max_accel: 3000
max_z_velocity: 25
max_z_accel: 30

# "RepRapDiscount 128x64 Full Graphic Smart Controller" type displays
[display]
lcd_type: st7920
cs_pin: ar16
sclk_pin: ar23
sid_pin: ar17
encoder_pins: ^!ar31, ^!ar33
click_pin: ^!ar35
`
[klippy.log](https://github.com/KevinOConnor/klipper/files/2544702/klippy.log)

KevinOConnor, 2018-11-07:
You haven't defined a [probe] section in your config.  A typical Voron printer uses the probe to perform automated bed levelling.  If you don't define the probe section then the code defaults to "manual probing" which would require you to manually jog the printer and perform the "paper test" at each test point (running NEXT when each point is at the correct height).

-Kevin

dgcaste, 2018-11-07:
I see! Thanks for the insight. I have a v1.5 which means the Z endstop is also the probe. Can the probe use the same switch?

KevinOConnor, 2018-11-07:
If the probe is used for the Z endstop, then configure the probe and set the Z endstop to use "probe:z_virtual_endstop" as described in config/example-extras.cfg.

Separately, I'm told the quad_gantry_level module performs better than the z_tilt module on the voron printers.

-Kevin

dgcaste, 2018-11-10:
That worked great. Thanks! I'm running a 1.5 which has two Z motors for the bed and not 4 for the gantry.

closed 2018-11-16 by KevinOConnor