Klipper3d/klipper #851: Bed tilt vs bed mesh

andremariz, 2018-11-03:
Wich one is better and whats the difference?

aeropic, 2018-11-03:
My config is as follows :
[stepper_z]
step_pin: PB3
dir_pin: !PB2
enable_pin: !PA5
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
position_min: -1
position_max: 240
homing_speed: 10

[probe]
pin: ^PC4
x_offset: -13
Y_offset: 22
z_offset: 2.30

Paulbsh, 2018-11-05:
Sorry for offtopic but I really think this is not an issue.
This is more suitable for a Klipper support group (like on facebook).
Putting this into the github issues section makes the issue section confusing.

aeropic, 2018-11-05:
Just decrease the z offset value

andremariz, 2018-11-05:
It is an issue as i stated before,  when mesh bed level finishes and it starts printing the nozzle is scrsping the bed. On the other hand, i had to set z min to - 1 becouse i was having an out of bounds error. So, if you can  help me figure this out fine, if this is not an issue for you cool, its for someone else.  Maybe the title is not the best i give you that. At the time i wasnt having this issue.

Arksine, 2018-11-05:
If the nozzle is touching the bed after calibration, you probe's z_offset is too large.  You need to decrease it.

andremariz, 2018-11-05:
Ok, i can print a line on the left side of the bed and then when it moves to the center to print the actual part its scrapin the bed! I can Z-offset in the printer menu but: 

1- Why isn´t it compensating if it did the mesh leveling

2- why can´t i sabe the offset that i set in the printer menu?

KevinOConnor, 2018-11-07:
If you are experience an issue then you'll need to attach the Klipper log file from the event - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

closed 2018-11-16 by KevinOConnor