Klipper3d/klipper #853: Support for MKS TFT

speendo, 2018-11-03:
At the moment, it seems like the MKS TFT displays (MKS TFT 3.2 and MKS TFT 2.8) are not supported by Klipper.

The MKS TFT 3.2 is connected to the MKS SBase at the AUX-1 port and runs its own firmware. I think it accesses the MKS SBase based on a protocol any other controller (like Octoprint) would use.

I would find it quite useful if it was possible to display at least some basic information on the MKS TFT (like temperatures and fan speed).

More features (like displaying the printing progress, controlling the printers movements or printing from the display's SD card) would of course also be welcome (but not that crucial)

megablue, 2018-11-04:
mks sbase intercept the serial communication and act as any proxy between the gcode sender and standard firmware but klipper doesn't send gcode to the controller board at all so... you cant "support" mks tft without finding a new way. it is highly impractical

speendo, 2018-11-04:
Thank you for your response.

So probably full support for MKS TFT is not possible without rewriting its firmware (there is a project that does that https://github.com/robotsrulz/MKS-TFT but it seems to be inactive) which I guess is unlikely to be done atm because there are more important features to implement in Klipper.

But wouldn't it be at least possible to make the MKS TFT display bed and nozzle temperature and PWM status (=fan speed) without "finding a new way"?

This would be the same functionality it has when printing from Octoprint with Smoothieware and would help me and other users to acquire some crucial information about the printer state without powering up another computer and looking into Octoprint.

sidddy, 2018-11-04:
Hi,

I'm currently also trying to figure out a way how to use an mks tft with my klipper/octoprint setup. Technically, I see three options:

1) Using standard mks tft firmware: with that firmware, the tft basically sends gcode commands via a serial connection on the AUX1 connector to the mainboard. Klipper currently does not read out that serial port at all, so you'd need to enable klipper to read these serial pins and somehow process the commands on the raspberry (potentially in klippy). 

2) Alternatively, you could "highjack" the aux1 serial connection of the standard firmware with another device (e.g. an esp8266) and send the commands of the tft via wifi to octoprint. Klipper would not be involved at all in that case.

3) the hard way: use a custom firmware which connects to octoprint. honestly, I did not like the standard firmware on the mks tft. So, I decided to follow the more difficult approach to re-implement the firmware on the mks tft and the mks tft wifi dongle to directly interact with octoprint. Klipper is again not directly involved, all communications go via octoprint. So far, I'm making progress following this approach and technically I do not see any more road blockers (tft, touch and wifi already works, woohooo). I haven't decided yet if it makes sense to share that project once I'm finished, since it's quite tailored for my printer setup & hardware...

speendo, 2018-11-05:
Wow @sidddy you seem to be very ambitious! I think nobody really likes the MKS firmware...

However, I don't have a wifi dongle and I don't really see a reason to get one... Why did you decide to leave klipper out of the game and use a different protocol (no offense, I just would consider this far more difficult).

Anyway, even if your project is probably not for me (I would possibly rather buy a cheap display than a wifi dongle) I would love to see your project progressing. I hope you find sense in sharing it soon!

By the way: is it possible to attach a different display than the MKS TFT to the MKS SBase?

sidddy, 2018-11-06:
@speendo: I decided to connect via Octoprint because I wanted to control the hole "start & monitor a print" process from the touchpad, without accessing the OctoPrint Web UI from a PC. So, I need access to the list of gcode files stored on Octoprint and trigger the job selection & execution. This is only possible via the Octoprint API, not (yet?) via klipper. Also, I have a delta printer running on a poor 8 bit atmega, so I didn't want to introduce any unnecessary load on the MCU for handling the serial connection.

The MKS Wifi dongle is a normal ESP8266, btw. You could connect any other cheap esp8266 as well, no need to use the mks-branded one.

Regarding other displays: It should be possible to connect any klipper-supported display to your board, but I haven't tried that yet with my Gen L (and hopefully won't need to).

KevinOConnor, 2018-11-30:
Certainly looks like an interesting thing to support in Klipper.  It doesn't look like anyone is actively working on it right now though, so I'm going to close this until it appears someone is actively working on it.

Separately, FWIW, it looks like these "MKS TFT" displays use an STM32F103 type micro-controller to update an SPI controlled display.  In theory, one could port Klipper directly to the micro-controller on the board.

-Kevin

closed 2018-11-30 by KevinOConnor

andreashb2, 2019-01-29:
@sidddy Hi. How are you doing on rewriting the firmware?

sidddy, 2019-01-29:
@andreashb2: My custom MKS TFT firmware (with wifi connectivity to octoprint, no direct klipper interaction) is meanwhile finished and I'm using it "productively" on my printer. It works quite well for me, but the coding would still need some polishing & installation documentation before I could release anything to public... Would you be interested to give it a try? I won't have any time to continue working on it before mid march unfortunately, but I could publish it on github around end of march probably...

manu7irl, 2019-01-30:
how did ou manage to connect your mcu to klipper wirelessly?

sidddy, 2019-01-30:
@andreashb2: I'm using the 3.2" version... I'm not sure if the 3.5" will be compatible. We can try to find out in March, unfortunately not right now (today is my last "online" day for quite a while...)

@manu7irl: I have the wifi extension for the MKS TFT board, which is basically an ESP8266 with a connector that fits directly on the MSK TFT. I flashed that ESP8266 with a firmware which acts as a proxy between the MKS TFT (connected via serial line to the ESP8266) and my Octoprint (connected via wifi & API). Therefore the display does not interact with klipper directly (except for "stealing" 5V power from klipper's mcu board for the display), but only via Octoprint.

manu7irl, 2019-01-30:
So your mcu is not connected to the wifi to serial but only the mks tft
screen? so the comands are passed through octoprint but not via klipper.
Am I right?

sidddy, 2019-01-30:
yes, correct

andreashb2, 2019-01-30:
@sidddy sure no problem! The only difference in the stock firmware is the preview of Gcodes directly on the screen. If its possible to implement it would have been nice, but it is not very necessary. Just reply to this thread when you are back in march.

fcollingwood, 2019-03-15:
@sidddy I'm keen to try this out too!

andreashb2, 2019-06-23:
@sidddy Been a few months now. How are you? I guess you haven't had that much time, but I am still very interesting in testing / helping out with the custom firmware:)

sidddy, 2019-07-12:
@andreashb2 I finally managed to get back to this project and also pushed it to github. There you go: https://github.com/sidddy/OctoScreen