Klipper3d/klipper #854: Fysetc TMC2208 - Error: Unable to read tmc2208 'extruder' register IFCNT

cassianoleal, 2018-11-04:
I'm trying to set up Fysetc v1.0 TMC2208 drivers in UART mode.

I have soldered the PDN pin to the top of the board and have bridged the 2 solder pads on the bottom of the PCM. I have tried setting the `uart_pin` to different pins on the MKS Gen L board but always get the same error. I have removed all the MS* jumpers from the board as well.

I don't know what else might be wrong. Sorry if this is not the best place for this issue but I'm out of ideas.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2545445/klippy.log)

Gomer05, 2018-11-04:
Hi cassianoleal, I have a MKS Gen L board with TMC2208's on X,Y and Z and Fysetc V1.0's.  All with the uart working.   
We had another case of the same symptoms a few weeks ago. See: https://github.com/KevinOConnor/klipper/issues/761 . In this case the issue seems to have been bad drivers chips.   Do you have another that you can try?  Your config looks good to me, but in my case I had originally wired up for Marlin, did some testing then found Klipper(and never went back), so my wiring is for Marlin.  It shouldn't make a difference.  As Kevin mentions in his docs Klipper should work with one wire.

MatejBosansky, 2018-11-05:
Hello guys.

Got same issue. MKS gen L with 4x TMC2208. After setting jumpers on MS1 and MS2 on extruder, problem has been solved.

cassianoleal, 2018-11-05:
@Gomer05 Did you have to do anything else on the drivers? Did you actually solder the jumper pads?

@MatejBosansky Do you mean you had to close MS1 and MS2 to get UART working?

MatejBosansky, 2018-11-06:
@cassianoleal

Yea but my problem probably was, that I had before closed jumper on MS3 and didnt tried pull out all jumpers. When I tried now pull out all jumpers, it is still working. So my issue was jumper on MS3. Sorry for misdirecting.

KevinOConnor, 2018-11-07:
@cassianoleal - your log indicates you are getting an "ADC out of range error" - make sure your thermistors are configured and working correctly before going on to configure the tmc2208.

-Kevin

cassianoleal, 2018-11-07:
@KevinOConnor I can. As soon as I comment out the `[tmc2208 extruder]` section I can print just fine, everything working as expected. When I put the section back in I get those errors.

KevinOConnor, 2018-11-07:
Are you saying you only get the "ADC out of range error" when the section is not present?  If so, can you provide the log from a session with the tmc2208 section not present, then add it back, issue a RESTART, and then attach the full log?

-Kevin

KevinOConnor, 2018-11-07:
Also, separately, make sure the motor power is enabled during testing (the tmc2208 itself is powered from the motor power supply, not the 5v mcu power supply).

-Kevin

cassianoleal, 2018-11-07:
@KevinOConnor Sorry, I think the log attached to the OP was from when I was testing just the UART in isolation on a spare board. I had the board powered from USB with just the one motor driver attached.

Please find the log after doing the procedure you requested. Motor is definitely powered, printer works as expected when the TMC section is not present.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2558341/klippy.log)

KevinOConnor, 2018-11-07:
Okay.  The "Unable to read tmc2208 'extruder' register IFCNT" occurs when the mcu is unable to communicate with the driver.  As above, make sure the motor power is on to ensure the tmc2208 is on.

I have Fysetc tmc2208 drivers in a Gen-L board myself.  They work fine for me.  I had to bridge the solder jumper between ms3 and the uart-pdn pin on each of the fysetc drivers.  I then removed all the jumpers from the gen-l board and used a jumper wire to connect the ms3 pin on the gen-l board (the pin connected to ms3 from one of the jumper pins under the driver) to a spare pin on the gen-l.  I used pins ar64, ar59, ar42, ar40, but there isn't anything special about those particular pins.

Beyond checking the wiring, I don't have any additional advice.

-Kevin

cassianoleal, 2018-11-07:
I have removed all MS jumpers from the board and checked that the jumper wire is good. I did that by using a multimeter to check for continuity between the solder blob and the loose end of the wire (the one that connects to the board).

I have connected the re-soldered PDN pin to D1 (AUX-1's top-right pin when looking at the board with the driver slots on top like on the diagram).

[image]

KevinOConnor, 2018-11-07:
Yes, if you can move the stepper then stepper motor power is enabled.  I have a fysetc v1.2 board so it looks a little different for me, but otherwise your wiring looks correct to me.  ar0/ar1 are used for mcu/host serial communication - use the pins on aux-2 instead.

-Kevin

cassianoleal, 2018-11-07:
Hah! `ar42` works, thanks a bunch!!! 😃 

```
Send: DUMP_TMC STEPPER=extruder
Recv: // GCONF:          000001c0
Recv: // GSTAT:          00000001
Recv: // IFCNT:          0000000c
Recv: // OTP_READ:       0000000a
Recv: // IOIN:           20000141
Recv: // FACTORY_CONF:   0000000a
Recv: // TSTEP:          000fffff
Recv: // MSCNT:          00000008
Recv: // MSCURACT:       00f7000c
Recv: // CHOPCONF:       14030053
Recv: // DRV_STATUS:     c0020000
Recv: // PWMCONF:        c80d0e24
Recv: // PWM_SCALE:      00000003
Recv: // PWM_AUTO:       000e0024
Recv: ok
```

closed 2018-11-07 by cassianoleal

manu7irl, 2018-11-08:
@KevinOConnor hi kevin fysetc is f6 mcu?
do you if it is the same layout or pinout as 1.3?

manu7irl, 2018-11-11:
> Yes, if you can move the stepper then stepper motor power is enabled. I have a fysetc v1.2 board so it looks a little different for me, but otherwise your wiring looks correct to me. ar0/ar1 are used for mcu/host serial communication - use the pins on aux-2 instead.
> 
> -Kevin
Hi I have the FYSETC F6 v1.3 do you know how can I easily get the layout? maybe it is close to the 1.2?