Klipper3d/klipper #856: [Q] what happend to filament sensor...[FR]G60 and some other stuff

Hywelmartin, 2018-11-04:
I belive that a filament sensor was on the way i to klipper... As I understood,  the intention was to let Octoprint handle everything in between the filament sensor trigger and the restart of the printing.. 
I think that it is better to build as many features into klipper as possible due to control over functions. And during "sd-printing" from lcd Octoprint doesn't know anything....

I have on the other hand some Ideas:
Implement
- G60: save current position to slot. or a klipper specific SAVE_POS/ SAVED_POS[x,y,z]
-[ifpin my_ifpin] that do some Gcode upon high or low with the possibiliti to active/deactivate thru Gcode
-and lastly the wait for click on some button(click pin on the rotary encoder or a my_ifpin

Then I believe that the rest would be possible to build with existing G-Code macros function..

_mock up-ish

[ifpin my_filsensor]
pin: ^ar99
#cycle: 
# how offen in msec default 2000msec
gcode: 
M600

[gcode_macro M600]
gcode:
SAVE_POS
G91
DEACTIVATE my_filsensor
G1 Z25  E-15 F500
G90
G1 X0 Y-90
G0 E-50 F1000
WAIT_CLICKPIN
G0 E50 F1000
G1 E50 F500
WAIT_CLICKPIN
ACTIVATE my_filsensor
G1 SAVED_POSXY F1000
G1 SAVED_POSXYZ F1000

manu7irl, 2018-11-04:
good idea but G1 E50 should replace with a variable you put in printer.cfg to respect the length needed for direct drive or bowden tube. I think something like max_load_fil min_load_fil

Hywelmartin, 2018-11-04:
You kind of put it in printer.cfg....

KevinOConnor, 2018-11-07:
FYI, there was a pull request for filament sensor support (pr #545).  From my perspective, the main item remaining was testing that everything works correctly during filament changes.

I'm not really sure what you are proposing, but as a high level comment I'd caution against "programming in g-code".  I suspect writing a python "extras" module (as described in https://github.com/KevinOConnor/klipper/blob/master/docs/Code_Overview.md ) would be more robust and much easier to maintain in the long term.

-Kevin

Hywelmartin, 2018-11-07:
@KevinOConnor  

My main goal is to get the functions of a filament sensor and a filament switching function aka M600. My main concern is that a lot of people would ask for many functions/settings/possibilities in order to be happy about it.

I can see a bunch of other applications for the individual function.

In Not at familiar with coding at any level... I have done some crude programs but...so this is over my head.

The #545 as far as I understood would sent a ping to Octoprint and let it take care of the actual filament change, I don't see that as a nice solution because that is out of the klipper scope(and frankly I only see Octoprint as a bunch of buttons on a screen)

 The functions i'm suggesting is:

-Possibility to save the current position  and get  the 3 different coordinates like G60
  SAVE_POS/ SAVED_POS[x,y,z]

-The function to wait until a user push a button preferable the CLICKPIN on the rotary encoder
  WAIT_CLICKPIN 

-The function of a Pin to spit out a Gcode if high or low, that could be activated/deactivated thru 
 terminal/extended gcode 
 DEACTIVATE my_ifpin

KevinOConnor, 2018-11-13:
Hi.  I understand.  Alas, I fear commands like WAIT_CLICKPIN and DEACTIVATE would be difficult to implement, difficult to maintain, and not provide a satisfactory user experience.  In particular, it would be very difficult to implement basic error handling with these types of commands.

You've mentioned you've done some programming in the past.  You might want to take a look at the filament sensor code in PR #545 - which is basically eleven lines of code:
```python
class FilamentRunoutSensor:
    def __init__(self, config):
        printer = config.get_printer()
        self.gcode = printer.lookup_object('gcode')
        buttons = printer.try_load_module(config, "buttons")
        buttons.register_button_press(config.get('pin'), self.runout_callback)
    def runout_callback(self, eventtime):
        self.gcode.respond_info("action:pause")

def load_config_prefix(config):
    return FilamentRunoutSensor(config)
```
I understand that not everyone is a software developer.  However, I think the above python code is only a little more complex to implement than multi-stage g-code scripts with Klipper specific commands.  Python has the further advantage that there are a large number of online tutorials and references available.

Cheers,
-Kevin

Hywelmartin, 2018-11-14:
@KevinOConnor my programming experience is very very basic, it has only been some if:s and timers. to automate some sequences that loops (boring and time consuming tasks at work).

I can understand your concerns. 

But I'm definitely not able to program this....

So I let go of my idea... but I still would like to see the functionality implemented(run out sensor and filament change)

andremariz, 2018-11-14:
I Haven run out filament and it stops the print but when resuming i got temperature error not heating in time .

KevinOConnor, 2018-11-23:
It doesn't look like anyone is currently working on this.  Lets move this to the existing #249 and #545.

closed 2018-11-23 by KevinOConnor