Klipper3d/klipper #857: Change Filament: Temporarily disable max_extrude_only_distance to change filament

speendo, 2018-11-04:
I can't believe I am the first to experience this, so probably I am doing something wrong. I am sorry. However, I don't know how to do this right.

I just wanted to change the filament in my bowden extruder so I issued

    G0 E-500

I got an error:

> Extrude only move too long (-500.000mm vs 50.000mm)

I understand that this could be disabled by issuing something like

    max_extrude_only_distance: 1000

In my settings file. However, this would disable a useful sanity check for my gcode... Is there a way to unload and load filament in a bowden extruder in a different way or should there be a way to temporarily disable `max_extrude_only_distance`?

Hywelmartin, 2018-11-04:
in your printer.cfg 

under 
[extruder]

add this and change the distance or make 10x G0 E-50

max_extrude_only_distance: 500.0
#   Maximum length (in mm of raw filament) that a retraction or
#   extrude-only move may have. If a retraction or extrude-only move
#   requests a distance greater than this value it will cause an error
#   to be returned. The default is 50mm.

speendo, 2018-11-04:
Thank you for your answer.

But, to be honest, I feel like both solutions are less than ideal.

Issuing `G0 E-50` 10 times is quite annoying for obvious reasons.

On the other hand, setting `max_extrude_only_distance` to 500 would disable a reasonable security measure.

Another ugly hack would be

    M109 200
    FORCE_MOVE STEPPER=extruder DISTANCE=-500 VELOCITY=1000
    G28

The best solution would be a way to temporarily disable `max_extrude_only_distance` or otherwise a command like `SET_MAX_EXTRUDE_ONLY_DISTANCE MAX_EXTRUDE_ONLY_DISTANCE=500`

like we have it for other values, e.g. `SET_PRESSURE_ADVANCE ADVANCE=0.000`

jakep82, 2018-11-04:
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg

speendo, 2018-11-04:
@jakep82 I had a gcode macro to perform firmware retractions. However, I made a little mistake and used `G82` instead of `M82` (what I intended to use).

Because of the reasonable setting for `max_extrude_only_distance`, my print was stopped in the first layer. Otherwise my print would have a big bowl of spaghetti and debugging the reason for that would have been very difficult.

The sanity check of `max_extrude_only_distance` makes sense in prints. However, it is impractical when it comes to filament loading/unloading.

speendo, 2018-11-04:
... and issuing the same command 10 times (whether it is in the console or in a gcode macro) is a bit ugly, isn't it?

jakep82, 2018-11-04:
The entire point of a macro is to issue multiple commands at once.  Even if you make max extrude a run time option, you'll still need to issue multiple commands to change it during load/unload, so you'll need a macro either way.

speendo, 2018-11-04:
I'm totally pro macro. But the macro should look something like

    [gcode_macro UNLOAD]
    gcode:
    G0 E-{DISTANCE} F3000
    default_parameter_DISTANCE: 500

and not like that
     [gcode_macro UNLOAD]
    gcode:
    G0 E-50 F3000
    G0 E-50 F3000
    G0 E-50 F3000
    G0 E-50 F3000
    G0 E-50 F3000
    G0 E-50 F3000
    G0 E-50 F3000
    G0 E-50 F3000
    G0 E-50 F3000
    G0 E-50 F3000

I hope you agree that the former is both: more elegant and more flexible.

jakep82, 2018-11-04:
I'm just giving you an option that works now, rather than something that isn't implemented (and might not be implemented). I personally don't care how ugly the macro is as long as it works.

ndupont-net, 2018-11-05:
Hi, from my experience a move of -50 won't work either as it is not shorter than the defined 50. 
I came to use 40mm steps in order to calibrate my extruder with a 200mm move.

Hywelmartin, 2018-11-05:
You can even make that macro look nice, with some different speeds and a short and positiv move to shape the end of the filament...

speendo, 2018-11-07:
@KevinOConnor Thanks for looking at it.

Although I accept your decision, what would be bad about making some more config settings changeable in the GCode terminal?

I would find that consistent with other commands like `SET_VELOCITY_LIMIT` or `SET_PRESSURE_ADVANCE`.

jakep82, 2018-11-07:
I'll give my 2 cents.  Velocity limit and pressure advance are printing parameters that affect print quality, and they need to be tuned.  Max extrude distance is a safety feature and has nothing to do with print quality.  I think it's a bad idea to give users the option to disable it because it creates the possibility of a catastrophic mistake.  Making a mistake with the value set at 500mm could cause an inconvenient mess to clean up.  Making a mistake with that safety feature disabled could lead to someone extruding an entire roll of filament and destroying a hotend (or worse).

jakep82, 2018-11-09:
As noted above, Klipper already provides a solution for you.  Simply increase your max extrude distance in printer.cfg to 100mm.

Nandox7, 2018-11-09:
Yes I have read that and it does not work.
Not close to my printer now but the reported value is negative so it does not match.
I have raised it as up as 500mm and it still wouldn't work.

jakep82, 2018-11-09:
Either you've encountered a bug, or you have a mistake in your printer.cfg.  Klipper doesn't care which direction you're extruding, only the absolute distance.  I would suggest opening a new issue, and posting your log file.

Nandox7, 2018-11-09:
Sure will do.

KevinOConnor, 2018-11-13:
> Although I accept your decision, what would be bad about making some more config settings changeable in the GCode terminal?

To be clear, I haven't made any particular decision.  If someone submits a pull-request I'll look at that request and make a judgement on the code in that particular request.

My high-level advice to developers and users is to use the existing mechanisms (gcode_macro/max_extrude_only_distance).  Both because it's available right now, and because I have a general preference to not add code and complexity to Klipper when one can effectively obtain the same results without adding additional code.

-Kevin

closed 2018-11-30 by KevinOConnor