Klipper3d/klipper #858: menu, deck/card not really equivalent, autorun

hg42, 2018-11-04:
these are some remarks fot the menu support, addressing @mcmatrix, because I don't have an email address.

Lately, I updated to @KevinOConnor's master branch, which also updated the menu support to include decks and cards etc.

I already had a quick menu before, much like a deck, with short entries with some commands and values that I can change in realtime.
I implemented it with the former menu system when decks were not available, like this:
```
[menu m__root]
type: list
name: Main menu
items:
  e__home, ___, e__kill, ___, e__restart, e__restart_all, ___, e__printing
  e__move_Z, ___, e__move_X, ___, e__move_Y
  e__bed_now, e__bed, ___, e__hotend_now, e__hotend, ___, e__fan
```
where e__home etc. are commands, and move, temperatures, etc. are values and realtime inputs. ___ is a simple menu item containing a space.

Now I played with decks and cards, mainly because a deck doesn't have a title (one more row for the quick menu) and it can "autorun" (an internal name, resulting in the given menu replacing the standard display page). A deck also has one more column on the left (no `>` or `*` marker).

In general I am pretty impressed by the flexibility of the menu system and I generally like the new deck/card approach.

Some observations/questions/suggestions:
* for the top level menu, the title could be omitted (at least if the name is empty or missing), making a menuj list more similar to a deck
* the menu marker could also be suppressed by a setting
* a menu list could be made very similar to a deck, if some items could be strings, like this:
    `"B",e__bed_now,"/",e__bed," T",e__hotend_now,"/",e__hotend," ",e__fan`
  Detection of the strings should be easy.
* is there another way to assign autorun to a menu? I would have expected a simple `autorun = True` in a menu section...
* why is autorun limited to decks? a menu list can also be used for autorun (tried that by disabling the check for "deck" in menu.py)
* the numbered items (`{0}`) are inconvenient to handle. For many changes I also need to change all numbers below. Names would be much easier. The numbers define the cursor sequence, which may allow a custom position independent sequence. For names the sequence would have been derived from the position of the interpolated items.
* for easier handling I tried to use a structure with one (combined) item per row:
```
[menu c__root]
type: card
name: Card
content:
    "{0}"
    "{1}"
    "{2}"
    "{3}"
items:
  e__printing
  e__home|e__kill|e__restart|e__restart_all
  c__pos_x|___|c__pos_y|___|c__pos_z|___|c__speed
  c__bed_now|_/_|c__bed|___|c__hotend0_now|_/_|c__hotend0|___|c__flow
```
* Unfortunately this didn't work. Apparently the `|` separator only works in alternate items. I can circumvent this restriction by adding a comma at the beginning of the row (`,a|b|c`), effectively alternating an empty text with the concatenation (which never switches to the empty text, which is nice).
* alternating texts don't accept commands or inputs, which prevents the row based solution
* I guess, `|`would be more suitable for alternation and comma for concatenation, but I understand why you did it this way

My ideal way to define a card would be like this:
```
[menu c__root]
type: card
name: Card
content:
  "{e__printing}"
  "{e__home}{e__kill}{e__restart}{e__restart_all}"
  "X{c__pos_x} Y{c__pos_y} Z{c__pos_z} {c__speed}%"
  "B{c__bed_now}/{c__bed} T{c__hotend0_now}/{c__hotend0} {c__flow}%"
```
or with some of my comments solved (omitted menu title, omitted marker column, autorun, string items):
```
[menu c__root]
type: list
name: Menu
items:
  e__printing
  e__home,e__kill,e__restart,e__restart_all
  "X",c__pos_x," Y",c__pos_y," Z",c__pos_z," ",c__speed,"%"
  "B",c__bed_now,"/",c__bed," T",c__hotend0_now,"/",c__hotend0," ",c__flow,"%"
```

The symbolic approach could be implemented by using something like this (when using `.format()`, from memory, not tested):
```
class InterpolationEvaluationException(KeyError):
    pass

class computed_dictionary(dict):
    def __getitem__(self, key):
        try:
            return dict.__getitem__(self, key)
        except KeyError:
            try:
                return do_something_with(self, key)
            except Exception, e:
                raise InterpolationEvaluationException(key, e)
...
  text_to_be_displayed = template.format(**computed_dictionary(dictionary_with_menuitems))
...
```

@mcmatrix I can start some work on this myself, but I wanted to ask for your thoughts, first.
If you want, I can try to explore and implement some of these things and create PRs either on Kevin's or your repo (though it may be difficult to switch the repo, because my printer needs some changes). I can also send patches.

hg42, 2018-11-04:
I already found a way to implement the deck with `{name}` instead of `{0}`
```
[menu c__root]
type: card
name: Card
content:
  "{e__printing}"
  "{e__home}{e__kill}{e__restart}{e__restart_all}"
  "X{c__pos_x} Y{c__pos_y} Z{c__pos_z} {c__speed}%"
  "B{c__bed_now}/{c__bed} T{c__hotend0_now}/{c__hotend0} {c__flow}%"
```

If no items are given, I simply replace the names with a count while collecting them in items.
```
class MenuGroup(MenuContainer):
    def __init__(self, manager, config, namespace='', sep=','):
        ...
        self.items = config.get('items', '')

class MenuCard(MenuGroup):
    def __init__(self, manager, config, namespace=''):
        super(MenuCard, self).__init__(manager, config, namespace)
        self.content = config.get('content')
        if not len(self.items):
            def collect_names(match):
                name = match.group(1)
                num = "{" + str(collect_names.count) + "}"
                collect_names.count+=1
                self.items += "\n" + name
                return num
            self.items = ''
            collect_names.count = 0
            self.content = re.sub(r'\{(.+?)\}', collect_names, self.content)
```

I first collected a list in self.items. Curiously this worked. But self.items seems to be a string.

KevinOConnor, 2018-11-07:
FYI, I'm not familiar enough with the menu code to provide detailed comments on much of the above.  (I'm hoping @mcmatrix will be able to comment.)

Separately, I agree that replacing the numbers with names would be an improvement.

mcmatrix, 2018-11-07:
If you look this example card then you see why there are index numbers not item names.

```ini
[menu __card2004_static]
type: card
name: Card 20x04
content:
    "\x00{0:3s}\x04\x7e{1:3s}\x04   {2}"
    "\x01{3:3s}\x04\x7e{4:3s}\x04   \xa5{8}"
    "\x07{6}  \x02{5} {7}"
    "{9}"
items:
    __card_hotend0_current
    __card_hotend0_target
    __card_zpos
    __card_hotbed_current
    __card_hotbed_target
    __card_frpeed
    __card_fnspeed
    5,__card_prt_time, __card_usb_progress, __card_sdcard_progress
    __card_status
    __card_msg,__card_xpos|__card_ypos|__card_epos
```

For example line 
` 5,__card_prt_time, __card_usb_progress, __card_sdcard_progress`
is Cycler, it will alternate items separated by comma after each 5s

line 
`__card_msg,__card_xpos|__card_ypos|__card_epos`
is also Cycler but without timer. It will show first enabled item.
`__card_xpos|__card_ypos|__card_epos` is grouping (merging) it as one item in Cycler.

Your menu card will be fine this way.
```ini
[menu c__root]
type: card
name: Card
content:
  "{0}"
  "{1}{2}{3}{4}"
  "X{5} Y{6} Z{7} {8}%"
  "B{9}/{10} T{11}/{12} {13}%"
items:
  e__printing
  e__home
  e__kill
  e__restart
  e__restart_all
  c__pos_x
  c__pos_y
  c__pos_z
  c__speed
  c__bed_now
  c__bed
  c__hotend0_now
  c__hotend0
  c__flow
```

I need to analyze this named approach little bit more.

mcmatrix, 2018-11-08:
>But after renumbering the template several times (and making mistakes), I thought it should be possible to make this easier.
>...
> It only transforms the named variables into the numbered syntax when there is no items: option.

I think this name replacement so that internally menu code is still using indexes is quite clever.
Thanks.

I need to analyze it and see how it will play with other menu features/functionalities.
If okay then I'll try to implement the best of it into the card functionality but I need to finish #741 first!
It already includes menu enhancements.

> btw. I think the cursor will jump around when turning the knob in __card2004_static, because it follows the numbers.

No, there's no edit items in that card.
But if you have edit items then the cursor will follow the menu items order in items attribute.
If you include items in content in same logical order then it should be ok.

Actually, the content doesn't know anything about the items, it'll just format n-lines of texts.
The menu system just gives to that format functionality an indexed list of strings (rendered items).

hg42, 2018-11-09:
> No, there's no edit items in that card

ah, thanks, that's why I didn't notice this, when using __card2004_static.
My menus were editable from the beginning, so when using cards I need proper numbering.

When I find some time, I can try to implement the more general syntax transformation. Shouldn't be too difficult (I think, but the devil is in the details).

mcmatrix, 2018-11-22:
@hg42 FYI

I'm experimenting with something like this. Havent yet fully tested it.
It should parse format, extract variables and put it together again.

```py
import string
...
class MenuCard(MenuGroup):
    def __init__(self, manager, config, namespace=''):
        super(MenuCard, self).__init__(manager, config, namespace)
        self.content = config.get('content')
        if not len(self.items):
            self.content = self._parse_content(self, self.content)

    def _parse_content(self, content):
        formatter = string.Formatter()
        out = ""
        items = []
        count = 0

        try:
            parsed_content = list(formatter.parse(content))
        except Exception:
            logging.exception("Card content parsing error")

        for part in parsed_content:
            out += part[0]
            if part[1]:
                out += "{%s%s%s}" % (
                    str(count),
                    ("!" + part[3]) if part[3] else '',
                    (":" + part[2]) if part[2] else '',
                )
                items.append(str(part[1]))
                count += 1

            self.items = "\n".join(items)
            return out
...

```

If this is okay then i'll add it as PR

mcmatrix, 2018-11-23:
Please check #923

mcmatrix, 2019-01-10:
This functionality was committed to master as https://github.com/KevinOConnor/klipper/commit/005cbe157ac74e79043a55cf31bd04e67f7e1707

closed 2019-01-21 by KevinOConnor