Klipper3d/klipper #859: printer is not ready

Logjm, 2018-11-05:
I have tried installing klipper a million times and I keep getting the printeer not ready command.  i would appreciate some trouble shooting advice.  Everything seems to be correct looking throught the files in filezilla.

bruce356, 2018-11-05:
Hi, if you want someone to help you need to attach your klippy.log file found in the /tmp/ folder.
I know it can be frustrating but once the errors are sorted you will be happy to be using Klipper.
regards - bruce

jakep82, 2018-11-05:
You need to attach your log file. 

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

Logjm, 2018-11-05:
The directions aren't clear enough.  I don't know how to retrieve the log.

BlackStump, 2018-11-05:
you need to give a little more info on how to better direct you.
are you using Windows OS to run Octoprint?
if so you need WinSCP to retrieve the log file mentioned above from you Rpi of OrangePi or any other linux  based single board computer that you are installing/running Klipper on.
https://winscp.net/eng/index.php
what controller board are you using etc etc?
 have you read and followed the installation instructions? have you looked at the config files that is right for your printer or controller because there can be hints there as well.

LazaroFilm, 2018-11-05:
To retrieve the klippy log, go into your Pi via FTP (or pull the as card and put it in a card reader) in the root of the card is a folder named tmp and inside should be a file named Klippy.

closed 2018-11-13 by KevinOConnor

thesleepydog, 2019-01-08:
I had this error. I re-flashed the compiled firmware for my MEGA 2560, using UART0 (didn't matter what port on the Pi the Mega is plugged into), and 25000 baud to my arduino mega and it worked