Klipper3d/klipper #861: Bed mesh calibrate - Move out of range

mbgroot, 2018-11-05:
Greetings!
When printing, I found a “lens” - in the center the distance from the nozzle to the table coincides, but to the edges it is smaller and the nozzle touches the table. I decided to increase the area for "bed mesh" and indicated the points 100,100 and -100,100 in the config. At the beginning of the calibration, I received an error:
`Move out of range: -100.000 -100.000 5.000 [0.000]`

Maybe I'm doing something wrong? How to properly remove the "lens"?

[klippy.log](https://github.com/KevinOConnor/klipper/files/2547333/klippy.log)

mbgroot, 2018-11-05:
Sorry, I understood why this error is displayed. I miscalculated the coordinates.
The question about the "lens" is relevant.
Thank.

Arksine, 2018-11-05:
This is because you have a delta printer, and a square mesh cannot cover a significant portion of a round bed.  

I'm working on a solution for this, but currently I don't have much time to dedicate toward it.  It may not be until December until I have the time to dedicate toward the fix, but it will come.

closed 2018-11-16 by KevinOConnor

klHaribo, 2019-01-26:
I found the same "problem"...
prepared logfiles, startet to write an issue.... ... and found this ;)
Do you have some news?

Arksine, 2019-01-27:
I do have some code that should be able to probe a radius on a round bed, I just haven't got around to testing it yet.  Right now I'm doing some filament sensor work, when I'm done I'll do some basic testing and submit a PR, where folks with delta printers can test it better as I myself do not have a delta.

KostteT, 2019-01-30:
Ready for testing!

Arksine, 2019-01-31:
Please see Pull Request #1177 if you are interesting in testing support for probing round beds.

klHaribo, 2019-01-31:
Here!
but I don't know how to install it. tried to copy the bed_mesh.py by hand.
but klipper says Option 'radius' is not valid in section 'bed_mesh'

Arksine, 2019-01-31:
@klHaribo Great, thanks for the feedback.