Klipper3d/klipper #862: How to run Klippy in simulation on linux?

eyal0, 2018-11-05:
One of the OctoPrint-PrintTimeGenius users wants to use Klippy: https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/79

To do this, there would need to be a way to take a gcode file and Klipper configuration and figure out how long each line of gcode will take to run in practice.

It seems like it should be pretty easy: Just run Klippy, have it dump to some test.serial file, and then process that.

I have not been able to run Klippy on my linux computer.  I tried these directions: https://github.com/KevinOConnor/klipper/blob/master/docs/Debugging.md but klippy failed:

```
python ./klippy/klippy.py test/configs/hostsimulator.config -i ~/Downloads/Toothpaste_Squeezer.gcode -o test.serial -v -d out/klipper.dict
```

```
INFO:root:Starting Klippy...
INFO:root:Start printer at Mon Nov  5 13:55:13 2018 (1541444113.7 16836.6)
ERROR:root:Config error
Traceback (most recent call last):
  File "./klippy/klippy.py", line 136, in _connect
    self._read_config()
  File "./klippy/klippy.py", line 119, in _read_config
    config = pconfig.read_main_config()
  File "/usr/local/google/home/eyalsoha/private/klipper/klippy/configfile.py", line 175, in read_main_config
    regular_config = self._build_config_wrapper(regular_data)
  File "/usr/local/google/home/eyalsoha/private/klipper/klippy/configfile.py", line 163, in _build_config_wrapper
    fileconfig.readfp(sfile)
  File "/usr/lib/python2.7/ConfigParser.py", line 324, in readfp
    self._read(fp, filename)
  File "/usr/lib/python2.7/ConfigParser.py", line 512, in _read
    raise MissingSectionHeaderError(fpname, lineno, line)
MissingSectionHeaderError: File contains no section headers.
file: <???>, line: 2
'CONFIG_MACH_SIMU=y\n'
DEBUG:root:Klipper state: Disconnect
```

KevinOConnor, 2018-11-07:
I think it would be great to be able to use the Klipper host software to provide accurate print time estimates.  It would also be extremely useful because it would allow common g-code errors to be identified prior to printing.  (If you do something like this, you might want to have a conversation with @mmone - as integration with the OctoKlipper plugin would also be interesting.)

As for the error above - the test/configs/hostsimulator.config is not a printer config file.  You would need to use the user's actual printer config file (assuming they are using klipper) to get an accurate estimate.  Alternatively, for testing purpose, you could use one of the printer config files in the config/ directory.

-Kevin

eyal0, 2018-11-07:
Thanks for the reply.  I tried a different config but it still didn't work:

```
└──> python ./klippy/klippy.py config/printer-wanhao-duplicator-6-2016.cfg  -i ~/Downloads/Toothpaste_Squeezer.gcode -o test.serial -v -d out/klipper.dict

INFO:root:Starting Klippy...
INFO:root:Start printer at Wed Nov  7 10:55:59 2018 (1541606159.7 178883.9)
INFO:root:Extruder max_extrude_ratio=0.266081
INFO:root:Loaded MCU 'mcu' 57 commands (v0.6.0-592-gf6c2e8b-20181105_133550-eyalsoha24.nyc.corp.google.com / gcc: (Debian 7.3.0-5) 7.3.0 binutils: (GNU Binutils for Debian) 2.30)
MCU 'mcu' config: RECEIVE_WINDOW=192 SERIAL_BAUD=250000 CLOCK_FREQ=20000000 SOFT_PWM_MAX=256 STATS_SUMSQ_BASE=256
ERROR:root:Protocol error
Traceback (most recent call last):
  File "./klippy/klippy.py", line 140, in _connect
    cb('connect')
  File "/usr/local/google/home/eyalsoha/private/klipper/klippy/mcu.py", line 768, in printer_state
    self._connect()
  File "/usr/local/google/home/eyalsoha/private/klipper/klippy/mcu.py", line 623, in _connect
    self._check_config()
  File "/usr/local/google/home/eyalsoha/private/klipper/klippy/mcu.py", line 573, in _check_config
    self._send_config(None)
  File "/usr/local/google/home/eyalsoha/private/klipper/klippy/mcu.py", line 526, in _send_config
    cb()
  File "/usr/local/google/home/eyalsoha/private/klipper/klippy/mcu.py", line 392, in _build_config
    mcu_adc_max = self._mcu.get_constant_float("ADC_MAX")
  File "/usr/local/google/home/eyalsoha/private/klipper/klippy/mcu.py", line 670, in get_constant_float
    return self._serial.msgparser.get_constant_float(name)
  File "/usr/local/google/home/eyalsoha/private/klipper/klippy/msgproto.py", line 342, in get_constant_float
    return self.get_constant(name, default, parser=float)
  File "/usr/local/google/home/eyalsoha/private/klipper/klippy/msgproto.py", line 334, in get_constant
    raise error("Firmware constant '%s' not found" % (name,))
error: Firmware constant 'ADC_MAX' not found
DEBUG:root:Klipper state: Disconnect
```

Maybe I didn't choose the right value from `make menuconfig`?

What should I try?  Can you provide a command that will create a valid `test.serial` output?  If so, I can start with the working example and modify to my needs.

My final output required for good print time estimation is a mapping from file location to time spent printing.  For example, at byte 123 out of 9999 of the gcode file, the time printing so far would be 42 minutes.  And at byte 345 out of 9999, the time printing so far would be, say, 78 minutes.  Etcetera.  If you can think of a better way to come up with it, I'm all ears!

KevinOConnor, 2018-11-07:
If you download and extract the dictionaries I uploaded in #830, then you should be able to run:
`../klippy-env/bin/python klippy/klippy.py config/example.cfg -i test/klippy/move.gcode -v -o test.serial -d local/atmega2560-16mhz.dict`

It would be a significant amount of work to parse the serial output to extract the print times.  I suspect it would be notably easier to add a new command line argument to klippy/klippy.py that causes the host code to write out the desired information directly.  For example, after running the above, you'll notice a line that contains "Max time of 5.121750" - this is the total print time of the job (not including homing or heating time) - it's produced in toolhead:motor_off() - it's reporting the value of the toolhead:print_time member variable.

-Kevin

KevinOConnor, 2018-11-23:
It looks like the original question (running the host code in batch mode) was answered, so I'm going to close this issue.

closed 2018-11-23 by KevinOConnor