Klipper3d/klipper #863: [Feature request] Support for kinematics with 4 or more steppers.

NordicGuy, 2018-11-05:
Good evening, day or morning, depending on your location of course. 

I've been working on a 3D printer concept, which essentially revolves around a CoreXZ and CoreYZ stage driven in parallel to produce cartesian motion. This means that I would need four independent steppers, plus a fifth for the extruder, as a minimum to drive my prototype printer.  
I would like to use Klipper for this purpose.

Looking into the kinematics files there seems to me to be a slight problem to this in the way Klipper is set up - Each motor seems to be assigned to a rail, the motion for which is then calculated, but only three rails are provided for motion; rail 0 though 2, for X, Y and Z with rail 3 being for the extruder.

For RepRapFirmware on the Duet boards it seems pretty straight forward as CoreXYUV kinematics are already supported, all you would need to do is rename and replace the formulas and you'd be more or less done. But for Klipper it's not immediately obvious to me how support for four or more motion steppers? 
Maybe someone with a better eye for Python kan help in this regard? 

Thank you

KevinOConnor, 2018-11-07:
The "rails" are internal to the kinematic classes, so you don't need to use them if they don't make sense.  The fundamental concept is that the kinematic class translates movement in cartesian space to movement on each individual stepper.  How it does that is up to that particular kinematics class.

I don't know what kinematics you are trying to accomplish so there's not much advice I can provide.  There's some info on porting to new kinematics at: https://github.com/KevinOConnor/klipper/blob/master/docs/Code_Overview.md .  If you'd like to see an example of kinematics without a "rail" then the work-polar-20180823 development branch may be of interest (in particular commit 312b6146).

-Kevin

eduncan911, 2020-07-28:
Thanks @NordicGuy for bringing this up, and thanks @KevinOConnor for the hint of hope! :)

After nearly two years since this was first asked, we now have Bill's (deckingman) CoreXYUVAB build and we are starting to see high-end CoreXYU and CoreXYUV builds with independent X carriages.  And now, Duet supports CoreXYUVAB within its firmware.

@KevinOConnor , could you point us in more of a direction about "porting new kinematics", which I'm assuming is replacing the rails classes?  To give a break down of some examples of what this alphabet soup means...

We know CoreXY.  The "XY" is the kinematics around the x and y gantry.  

Now, imagine if you had a 2nd gantry mounted above the first.  In Deckingman's build, his heavy 1kg hotend-only (of 6x inputs) operates on the standard CoreXY with precision.  He built a 2nd gantry ontop of the first that carries the 4kg weight of the extruders (so it's less precise).  That second gantry "mirrors" the first XY kinematics; however, it still needs to "home" independently of the XY gantry.  Therefore, the Duet guys created the UV kinematics to focus on homing the UV gantry (the extruders gantry).  

Then his massively tall machine started to rock back and forth printing at 300mm and 400mm speeds.  To address this, he built yet a 3rd taller level gantry - at the very top of the machine (6 feet tall now) that simply carries the conuter-weights that offsets the XY and UV gantries. This gantry continues to mirror the original XY kinematics; however, in reverse order this time (so to counteract opposing forces).  And, it also needs to home independently.  So the Duet guys created yet a 3rd set of XY kinematics, called the AB rails.

So this is where you get 3 different gantries, XY UV AB for the CoreXYUVAB setups.

---

If that wasn't enough, then you have other multiple-kinematics with insane setups such as the [BCN3D SIGMAX R19](https://www.bcn3d.com/bcn3d-sigmax-r19/) that uses normal Cartesian and/or CoreXY kinematics (not sure which).  However, they added a 2nd X mount/hotend that moves independently of the first X mount.  They call it something like the Independent Dual Extruder (IDEX) (I'm sure they have a trademark somewhere for that, lol).  

It's pretty wild though.  You can run it in "Copy" mode, that allows you to print two objects at the same time.  They have a "Mirror" mode that lets you print two objects at the same time like copy mode, but in a mirror of each other.  

And finally, you can have tool-changer support that allows you to switch between extruders much like normal dual hotends today.  However, this requires a 3rd "U" rail system, e.g. CoreXYU, to allow you to address those 2nd independent X movements now called a "U" rail.

**Where do we start with Klipper to support these, and other custom functionalities?**  :)  I personally have a Delta, and a CR-10 I can start to tinker with.  I also have a Dremel 3D20, that Klipper already has the ARM support of.  However, I need to figure out the bootloader and pinouts.