Klipper3d/klipper #864: [REQ] Klipper Native Port accessibility Arduino DUE

bruce356, 2018-11-06:
Hi, once the Klipper firmware is flashed to the Arduino Due via the Programming port, it seems that Klipper does not support accessing the Native port.
Would it be possible to include programming in the firmware loaded to the Arduino Due so that the Native port becomes accessable.

The Configuration is a RPi3 with host software Klipper running with Octopi. There is no problem in flashing to the Arduino DUE through the Programming port and afterwards connecting through the Programming port (ttyUSB0) but a reset is required (clone DUE).

Being able to connect through the Native port would eliminate having to press the reset button. But because Klipper (I assume as I am not a programmer) does not have  provision or support for accessing the Native port it is not posible. This is confirmed (my limited knowledge) by the fact that if the Native port is connected to a windows computer (after flashing Klipper), windows does not recognise that this port exists (no response at all) but does recognise the Programming port.

I know Marlin functioning is different on the Due, but it has the option to activate both (or not) ports, and when Native port is plugged into a windows computer with Marlin loaded, the port is recognised and connected (without having to press the reset button). Connecting the Programming port also requires a reset with Marlin.

Sorry for the long post - thanks and regards - bruce

jakep82, 2018-11-06:
This is a hardware issue. I have a genuine Due, and the programming port works fine. I also have a clone Due, and the only way I could make it work without pressing reset was to add a resistor on one of the fets. This  isn't something that can fixed in software. New genuine boards have a design revision that solves the issue. If you aren't comfortable soldering a tiny SMD resistor on the board, then I would recommend spending the extra money for a genuine board ($37.44 right now on Amazon)

https://copperhilltech.com/blog/arduino-due-design-flaw-due-wont-start-after-poweroffon-requires-reset/

bruce356, 2018-11-06:
@jakep82 thanks for the reply, I was aware of the reset problem with the programming port on clones (its a very small fet & hard for me to try and to solder a resistor), that is why I was asking for a solution (if possible) regarding the Native Port, I thought if it can be done in Marlin it could possibly be done in Klipper, if the dev or someone else was so inclined.
Regards - Bruce

jakep82, 2018-11-06:
It's not related to programming or native ports.  The code on the MCU will not run until reset is pressed on clone boards.  This can't be fixed in software.

bruce356, 2018-11-06:
@jakep82 , I feel you are wrong, with Marlin loaded and both ports active in firmware, to connect through the Programming port using Pronterface one needs to perform a reset before Pronterface can connect through the port.

With the same configuration as above to connect through the Native port using Pronterface NO reset is required and Pronterface will connect each and every time.
Regards - Bruce

KevinOConnor, 2018-11-07:
Using the native port would require writing a USB driver for the SAM3X8E chip.  Writing a USB driver is a significant amount of work.  I've done it for several other chips, but doing it for the SAM3X8E isn't a priority for me as the Due has a readily available USB to serial adapter that generally works fine.

-Kevin

bruce356, 2018-11-08:
@KevinOConnor thanks for the reply, USB to serial adaptor might be a bit messy for this installation, I will persevere with the Programming port until someone (if ever) writes a USB driver.

@jakep82  I see you managed to get your  "RepRapDiscount Full Graphic Smart Controller" working.
I am not sure if you have the RAMPS-FD with LCD adapter (as I do) or if you connected directly to the Due pins.

I would like to get the display working if possible, any guidance would be appreciated.
My config is is below:-

```
[display]
lcd_type: st7920
cs_pin: ar16		;LCD RS
sclk_pin: ar23		;LCD D4
sid_pin: ar17		;LCD ENABLE
menu_timeout: 60
menu_root: __main
encoder_pins: ^ar33, ^ar31	;BTN_EN1 & BTN_EN2
click_pin: ^!ar35			;BTN_ENC

```
Regards - bruce

jakep82, 2018-11-08:
I'm using it connected directly the Due pins.  I took a quick look, and it seems you have things connected correctly.  The only advice I can offer is try connecting it without the adapter.  Also make sure you don't have any errors in your config.  I had a minor fan config error, and nothing would display until I fixed it.

KevinOConnor, 2018-11-23:
It doesn't look like anyone is actively developing this right now, so I'm going to close it.  I'll reopen if someone starts working on it.

-Kevin

closed 2018-11-23 by KevinOConnor

KevinOConnor, 2019-02-18:
FYI, should be implemented now.

-Kevin