Klipper3d/klipper #865: MCU shutdown - error message suggestion (Pressure Advance)

jameswood, 2018-11-06:
I've spent several hours chasing down an error that causes the mcu to crash on two different printers. It appears to be related to Pressure Advance. The error is `MCU 'mcu' shutdown: Rescheduled timer in the past`

Here is the most-minimal gcode I've come up with that causes the problem:

```
G28 ; home all
G90 ; xyz absolute
M82 ; extruder absolute 
SET_PRESSURE_ADVANCE ADVANCE=0.6
G92 E0 ; zero extruder
G1 X4.422 Y4.100 Z0.2 F800 ; ok
G1 X4.506 Y4.357 E0.0934 ; CRASHY TIMES
```

Dropping `ADVANCE` to `0.5` prevents the crash for this particular move on these particular printers (Titan extruders).

It appears that the extrusion of 0.0934mm is causing the extruder to need more steps than the MCU is capable of delivering. This raises two questions:

1. Can the max step-rate of each stepper be limited somehow?
2. Could the error message be more reflective of the issue?

I've looked through the extruder settings in example.cfg and while `max_extrude_cross_section`, `max_extrude_only_velocity` and `#max_extrude_only_accel` are close to what I'm talking about, they're not quite there. `max_extrude_cross_section` throws an error but doesn't prevent anything, and the other two only apply to extrude-only moves. I think I'm asking for something more like a `max_steps_per_second` setting for each stepper driver. It would be great to be able to take advantage of the quality improvements made possible by Pressure Advance, without having to limit it to strange jumps that the slicer (in this case, Simplify3D) might make between infill and outer walls, which is I think where this snippet of gcode came from (happy to attach full print file if that helps).

Anyway, this problem took me about a day to track down, so I thought I'd throw the conversation open to the group. Is there any way we could we make this problem go away, or perhaps make it easier to identify?

The extended Octoprint error, below, isn't very helpful in explaining that the problem is caused by the extruder, and that it's due to pressure advance requiring a far higher number of steps than an ordinary move might need, even at low print speeds (here, 30mm/sec). Perhaps this message could include which axis it was that tripped the error, and if it's an E axis maybe it could mention Pressure Advance?

```
Recv: // MCU 'mcu' shutdown: Rescheduled timer in the past
Recv: // This generally occurs when the micro-controller has been
Recv: // requested to step at a rate higher than it is capable of
Recv: // obtaining.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! MCU 'mcu' shutdown: Rescheduled timer in the past
```

[long_klippy.log](https://github.com/KevinOConnor/klipper/files/2551446/long_klippy.log)

KevinOConnor, 2018-11-08:
Unfortunately, you've hit a bug in Klipper.  The Klipper code has a check to permit small extrusions that exceed the max_cross_section_check if the total amount extruded is tiny.  Unfortunately, that check is not working properly with pressure advance - the pressure advance code is amplifying that tiny extrusion in an attempt to account for pressure.

I'll take a look at a fix for it.

-Kevin

derdigge, 2018-11-15:
stombled into the same, is there an workarround till a fix is made??

manu7irl, 2018-11-15:
got the same problem also...

KevinOConnor, 2018-11-15:
@derdigge , @manu7irl - you'll need to attach the Klipper log file from the event.  It's unlikely you've hit this particular issue.  I recommend you each open new github issues with the relevant log.

-Kevin

manu7irl, 2018-11-15:
actually I think I found the source of the problem.
but if the problem returns I will send my log.

KevinOConnor, 2018-11-30:
As an update, I have looked at this issue.  Alas, it does not appear to be a simple fix.  I'm inclined to leave the code as is for the Klipper 0.7.0 release and look at possible solutions afterwards.

There are two main contributing factors to this issue.  The first is that Simplify3D is generating a strange extrude request - "G1 X4.506 Y4.357 E0.0934".  Which is a move of .513864mm while extruding 0.0934mm of filament.  That results in an extrusion of .224653mm^3, which is an extrusion cross-section of .437184mm^2, which would be an extrusion width of 1.5mm if one were using a layer height of .3mm.  I suspect this is just a bogus request from Simplify3D.  It would normally result in Klipper raising a maximum extrusion cross section error, but Klipper permits it because the total amount extruded (.0934mm) is tiny.

The second contributing factor is that the printer hardware is unable to execute enough extruder steps to obtain the configured pressure advance value for an extrusion that is of the max_extrude_cross_section area.  (The configuration of this particular printer has a high pressure_advance setting.)  When Klipper gets the strange movement request above, it caps the pressure advance to be no greater than what's configured as what a max_extrude_cross_section extrusion would have.  And, once it does that, it results in a request to the micro-controller which requires too many steps for it to handle.  Arguably, it would be better for Klipper to catch moves like the above and just completely disable pressure advance for them.  Alas, though, this would not really fix the issue as it's entirely possible that Simplify3D might produce a "questionable" move with an extrusion cross section just under the max_extrude_cross_section check and the code would honour that and produce a similar failure.

Separately, the original report raised the point that the error messages were not particularly friendly.  I agree with this assessment.  Alas, generating good debug messages during an error caught my the micro-controller is not easy.  At the point the error is detected, the micro-controller only knows that it could not keep up with the schedule of events requested of it - it doesn't know what part of the schedule was the root cause of the problem.  Ideally the fix is to catch these problems in the host code before transmitting them to the micro-controller.

-Kevin

drphil3d, 2019-01-24:
This sounds very similar to the issue I have been running into. Simplify 3d causing strange errors with the gcode it generates.

MarcinKnD, 2019-07-25:
I've also had a problem with me, three prints have been interrupted by stopping the printer with this message

Recv: // MCU 'mcu' shutdown: Rescheduled timer in the past
Recv: // This generally occurs when the micro-controller has been
Recv: // requested to step at a higher rate than it is capable of
Recv: // obtaining.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! MCU 'mcu' shutdown: Rescheduled timer in the past

last about 19.00 - 20.00 even with pressure_advance off after about 20%

what to do?

[klippy.log](https://github.com/KevinOConnor/klipper/files/3430576/klippy.log)

KevinOConnor, 2019-11-21:
With the merge of PR #1997, I believe this issue should be fixed.

-Kevin

closed 2019-12-23 by KevinOConnor