Klipper3d/klipper #867: Process to Remove/Uninstall Klipper from the Pi?

talldonkey, 2018-11-06:
Hi - Can you advise on the best way to cleanly uninstall Klipper? I'm not able to find it in the docs or online so far.

AxMod3DPrint, 2018-11-07:
There's nothing documented at the moment, however, I'll take a look at
writing a script to do so when I have 5 minutes.

Hywelmartin, 2018-11-07:
in the mean time you only have to stop the service and use octiprint as normal

KevinOConnor, 2018-11-17:
@AxMod3DPrint  - did you make any progress on that?

Thanks,
-Kevin

AxMod3DPrint, 2018-11-18:
Kevin, about the only thing I've found is to RESTART after setting the
correct pin state to kick the drivers into action.

KevinOConnor, 2018-11-23:
@AxMod3DPrint - earlier in this thread you mentioned you might take a look at writing an uninstall script.  Did you make any progress on that?

Thanks,
-Kevin

AxMod3DPrint, 2018-12-09:
> It doesn't look like anyone is working on this. At a high-level the steps
> would be to remove the files installed by scripts/install-octopi.sh.
> Probably the simplest way to do that is to reinstall octopi. Short of that,
> one could probably run sudo systemctl disable klipper to just not run
> Klipper at startup.
>
> I'm going to close this for now as it appears inactive.

Not yet, it's on the "to do" list.

KevinOConnor, 2019-02-18:
FYI, should be implemented now.

-Kevin