Klipper3d/klipper #868: Clarification on probe offset direction

jasonmhite, 2018-11-06:
This is mostly a request for clarification of the phrasing of the config description related to offsets when configuring a probe.The descriptions in `example-extras.cfg` say these offsets are "The distance (in mm) between the probe and the nozzle along the x/y/z-axis". I want to clarify, what are these offsets relative to?

Are they the distance from the probe location to the nozzle or from the nozzle to the probe? In other words, what is the sign convention for these offsets?

cosminr86, 2018-11-06:
it is relative to your 0 position. if your probe is towards 0 position, is with a negative sign, and vice versa

jasonmhite, 2018-11-06:
Perfect, thanks!

closed 2018-11-06 by jasonmhite

andremariz, 2018-11-06:
And what does it do? Becouse when we are setting points on a tilt point we have to give room to the probe becouse it wont calcular bases on this offsets