Klipper3d/klipper #870: pressure advance with TMC2208

Jeroen0553, 2018-11-07:
Printer info:
Anycubic kossel linear plus
running tmc2208 on all axis including extruder (with active cooling)
pressure advance set to 0.700 and pressure advance look ahead time to 0.020

I started tuning pressure advance and when I was finished put in the values I found worked best for me and restarted the firmware. Then I did some test prints with a xyz cube. I ran these at quite a high speed (nothing klipper wouldn't be able to handle I think), 100mm/s. However, after a few layers, the extruder stopped working, it didn't do anything anymore. After stopping the print and extruding via control tab in octoprint or starting a new print the extruder worked just fine again. It doesn't happen at the same spot every time, but quite soon after starting the print.
Any idea what might be causing this?

[work_directory.zip](https://github.com/KevinOConnor/klipper/files/2559440/work_directory.zip)

KevinOConnor, 2018-11-10:
I suspect it's the same reason discussed in issue #196 (the tmc driver incorrectly increasing "voltage" on a change of direction).  It seems the tmc2208s don't work well in standalone mode with stealthchop.  If you can enable the "uart" line and route it to a free pin on the micro-controller, the general feedback is that they do then work well on Klipper.

-Kevin

KevinOConnor, 2018-11-16:
I'm going to close this in favour of the existing issue #196.

closed 2018-11-16 by KevinOConnor