Klipper3d/klipper #871: Tool change does not work properly at Dual X system

3dgege, 2018-11-08:
Tool change does not work properly on my Dual X system
Extruder 0 and 1 work well. With a double pressure, the extruder is changed, not the tool head.
so promotes in a change of the second extruder filament butonly the first head move . if I switch manually just go with command SET_DUAL_CARRIAGE CARRIAGE = 1. simpify but take T0 or T1. ignored Klippy but with manual input
sorry for my bad english ;-)
best regards
Gerhard

# Support for cartesian printers with dual carriages on a single
# axis. The active carriage is set via the SET_DUAL_CARRIAGE extended
# g-code command. The "SET_DUAL_CARRIAGE CARRIAGE=1" command will
# activate the carriage defined in this section (CARRIAGE=0 will
# return activation to the primary carriage). Dual carriage support is
# typically combined with extra extruders - use the SET_DUAL_CARRIAGE
# command in the activate_gcode / deactivate_gcode section of the
# appropriate extruder. Be sure to also use that mechanism to park the
# carriages during deactivation.
[dual_carriage]
axis:x
#   The axis this extra carriage is on (either x or y). This parameter
#   must be provided.
step_pin: ar29
dir_pin: ar28
enable_pin:!ar39
step_distance: .0124
endstop_pin:^!ar36
position_endstop:469.3
position_min:60
position_max:469.3
#   See the example.cfg for the definition of the above parameters.

bruce356, 2018-11-09:
@3dgege , hi Gerhard sorry but I do not understand what you are trying to say, perhaps others can?
I have and IDEX printer using a RUMBA board and tool change works fine with T0 & T1.
Not sure if you are German but try explaining in your native language.
Also if you want help you need to attach your "klippy.log" file. Use the program "WinSCP" 
https://winscp.net/eng/index.php?#utm_source=winscp&utm_medium=app&utm_campaign=5.13.3
Regards - bruce

3dgege, 2018-11-10:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2568739/klippy.log)

3dgege, 2018-11-10:
Hello, I am happy, the printer is running. We found the mistakes on our FB group. the new design so that it will work in append. Thank you for your help. Greetings Gerhard
[klippy.log](https://github.com/KevinOConnor/klipper/files/2568797/klippy.log)

closed 2018-11-16 by KevinOConnor