Klipper3d/klipper #874: Heater error message is repeated several times

fritzw, 2018-11-08:
I just changed my bed heater from watermark to PID (not sure if this is relevant).

After replacing the cold glass plate on my already hot bed, naturally the temperature dropped, klipper complained: `Heater heater_bed not heating at expected rate` and shut down the printer.

So far, so good, safety features are working. Only it repeated the exact same message 58 times (see attached log file). This sounds like a bug, albeit a harmless one, so I'm reporting it.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2563921/klippy.log)

manu7irl, 2018-11-09:
did you try the calibrate pid function?

fritzw, 2018-11-11:
Ah, all right. If that is the desired behaviour, I guess we can close this issue.

closed 2018-11-12 by fritzw