Klipper3d/klipper #875: Adjust motor current

cho934, 2018-11-09:
Hi,

I succeeded to configure klipper with my printrboard (printrbot metal plus), very nice work !

My questions are the following:
-  about increasing the current in the stepper motors, Is it possible with the klipper firmware ?
With Marlin, if you can enter in an M909 to view the current value, M907 and an axis variable then the value as a percentage of current to set a new current value and an M910 to save that value to the board.

- what is the maximum speed of a common 3D printer we can test with klipper? 300mm/s? 400mm/s?

Have a nice day,
Christophe

FHeilmann, 2018-11-09:
From a quick google search the Printrboard uses an MCP4728 for stepper adjustment which is similar, but unfortunately not equal to the MCP4451 supported by klipper, so some dev work is needed to get them working.

As for speed, check out the maximum step rates of the microcontroller of your board hereL https://github.com/KevinOConnor/klipper/blob/master/docs/Features.md
and multiply that by your step distance to get the maximum speed.

KevinOConnor, 2018-11-24:
As a result of recent changes in Klipper, it should not be difficult to add support for the MCP4728 chip.  Let us know if you are willing to run tests with some experimental code changes.

-Kevin

KevinOConnor, 2018-11-25:
There is test code up on the work-mcp4728-20181124 branch.

cd ~/klipper ; git fetch ; git checkout origin/work-mcp4728-20181124 ; sudo service klipper stop ; make menuconfig ; make ; make flash ; sudo service klipper start

Then add the following to your printer.cfg file:
```ini
[mcp4728 stepper_current_dac]
scale: 2.327
channel_a: 1.3 # Extruder
channel_b: 1.1 # stepper_z
channel_c: 1.1 # stepper_y
channel_d: 1.1 # stepper_x
```

Run RESTART and verify that the given stepper current is applied to the stepper.  Try setting different currents in the config file (running RESTART after each change) and verify that they take effect.

-Kevin

KevinOConnor, 2018-11-30:
Any luck with the above?

-Kevin

cho934, 2018-12-28:
Working on it.

I'm finally available to do it !

cho934, 2018-12-28:
ok. To upload it to my printrboard, I need to follow that instead of "make flash":
https://reprap.org/wiki/Printrboard#Loading_Firmware_.28Windows.29

KevinOConnor, 2018-12-28:
Follow the steps above.  But, if "make flash" doesn't work for you, use the steps you've previously used to flash your board.

-Kevin

cho934, 2018-12-29:
Test successfully done.
Firmware version: v0.7.0-1-g1e157f5  
I tested reducing the current to 0.001 and see if the motors don't move. 
Then I printed a new object successfully with the normal values.

What the maximum current we can put ? (for the extruder for example ?)

KevinOConnor, 2018-12-30:
Great.  I'll look to merge this later in the week.

The max current depends on the driver and the motor.  Most drivers can't go over 2amps, and things get hot if one goes up to the max supported amps.  Most configs I've seen set the amps around a value of 1.

-Kevin

KevinOConnor, 2019-01-03:
FYI, this support was merged (commit c2086796).

-Kevin

closed 2019-01-10 by KevinOConnor