Klipper3d/klipper #878: Install to mks robin howto

userosos, 2018-11-09:
Who installed klipper on mks robin 2? Does it work well? Need instruction. How to do it, please.

jakep82, 2018-11-09:
That's a closed source board, and I've read it has a locked bootloader.  If you haven't already bought the board, don't.

KevinOConnor, 2018-11-23:
It doesn't look like anyone has information on how to run Klipper on the MKS Robin, so I'm going to close this.

closed 2018-11-23 by KevinOConnor

userosos, 2018-12-15:
Hi! Mks robin's bootloader was hack.
https://github.com/jmz52/MKS-Robin-Bootloader
Steps in the site (Russian language):
http://3dtoday.ru/blogs/jmz/mks-robin-part-two-the-taming-of-the-shrew-firmware/ 
Secret Key: A3BDAD0D4111BB8DDC802DD0D2C49B1E26EBE3334A15E40AB3B13C93BBAFF73E
Will it help for the problem?

KevinOConnor, 2018-12-19:
If I'm reading the above correctly, four steps would be required:
- update Klipper to support UART3 for communication
- update Klipper to support a 28KB bootloader offset
- the resulting out/klipper.bin would need to have the XOR applied to it
- create a Klipper config file

It doesn't look difficult.  But, why help Makerbase sell more of these boards?  It's not something I personally am planning to work on.

-Kevin

KevinOConnor, 2020-03-26:
FYI, it should now be possible to run Klipper on the "Robin" boards.  If interested, see the config file in commit cb414ae9 as a starting point.

-Kevin