Klipper3d/klipper #881: Mesh_bed don´t push Bltouch Pin

Ahrenfelde, 2018-11-10:
Hello,
when I am do a bed_mesh_calibrate the Pin from my Bltouch do not go out.
I can move the pin by servo commands in the terminal. When i enable the Bltouch per Terminal befor a bed_mesh_calibrate the Printer do not reacts to a contact with the Push pin.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2568358/klippy.log)
[printer.txt](https://github.com/KevinOConnor/klipper/files/2568359/printer.txt)

BlackStump, 2018-11-10:
I have a BLTouch that works just fine with bed_mesh_calibrate tho I removed
SET_SERVO SERVO=bltouch ANGLE=60        in the activation code as it is a test mode for M119 or query_endstop, how it works is you set angle to 60 in the terminal then manually press the pin up then type M119 in the terminal and a Triggered response should return,
I really dont know why SET_SERVO SERVO=bltouch ANGLE=60 was included in the example config.
I should note I have a genuine BLTouch not a clone! there is plenty of info about issues and peculiarities with clones that dont work as expected.

Ahrenfelde, 2018-11-10:
This works for me sadly not. I test varios versions of activation start code, but nothing works. It seems the activation startcode is just skipping here.
[printer.txt](https://github.com/KevinOConnor/klipper/files/2568836/printer.txt)

Ahrenfelde, 2018-11-10:
year no Problem, thanks!
You use the Bltouch to home your Z-axis?
I think the best way is, when i am resetting all. Start with a fresh Octoprint and Klipper from scratch.
I reply this issue when i am ready with this work, maybe tomorrow morning.

BlackStump, 2018-11-10:
yes I thought it would be easier to use the bltouch for Z homing as well.

ndupont-net, 2018-11-11:
Making it work took me a bunch of hours.

Here are the details that make it work : 
`[stepper_z]
...
endstop_pin: probe:z_virtual_endstop
`

Ahrenfelde, 2018-11-11:
i am actually in progress to build my klipper+octopi new.

I want to use the Bltouch only for Probing and home my Z axis to Z-Max with and microswitch. I had tested to home the Z axis with the Bltouch, but this won´t work for me yesterday.

Ahrenfelde, 2018-11-11:
Okay, it works now.
But only when i am using the Bltouch for homing and for Probing.
This is not exactly what i need. But for now this issue is solved i think. I will close this issue and open a new one with the other topic to split the homing and probing sensor.

Thanks for your help!

closed 2018-11-11 by Ahrenfelde