Klipper3d/klipper #883: TMC2130 config problem

Kowy54, 2018-11-11:
With my config for TMC2130 i´m getting error (Reported error: No section: 'stepper_x')
My config for stepper_x:
[tmc2130 stepper_x]
driver: TMC2130
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: ^ar3
#endstop_pin: ^ar2
position_endstop: 0
position_max: 220
position_min: -23
homing_speed: 50
cs_pin: ar40
microsteps: 16
interpolate: True
run_current: .650
hold_current: .500
sense_resistor: 0.110
stealthchop_threshold: 50
driver_IHOLDDELAY: 8
driver_TPOWERDOWN: 0
driver_BLANK_TIME_SELECT: 1
driver_TOFF: 4
driver_HEND: 7
driver_HSTRT: 0
driver_PWM_AUTOSCALE: True
driver_PWM_FREQ: 1
driver_PWM_GRAD: 4
driver_PWM_AMPL: 128
driver_SGT: 1

BlackStump, 2018-11-11:
you are missing stepper_x and y and z
[klippy.txt](https://github.com/KevinOConnor/klipper/files/2569303/klippy.txt)

[stepper_x]
step_pin = ar54
dir_pin = ar55
enable_pin = !ar38
step_distance = .0125
endstop_pin = ^ar3
position_endstop = 0
position_max = 220
position_min = -23
homing_speed = 50

[stepper_y]
step_pin = ar60
dir_pin = !ar61
enable_pin = !ar56
step_distance = .0125
endstop_pin = ^ar14
position_endstop = 0
position_max = 210
position_min = 0
homing_speed = 50

[stepper_z]
step_pin = ar46
dir_pin = ar48
enable_pin = !ar62
step_distance = .0025
endstop_pin = ^ar18
position_endstop = 0.5
position_max = 220

Kowy54, 2018-11-11:
Yeah, its working. :)

closed 2018-11-11 by Kowy54