Klipper3d/klipper #884: st7920 display glitching

voodoo-bravo, 2018-11-11:
Hello, 

I'm using a 12864 Full graphics display (identical to the RR-Discount one) with a RADDS 1.0 on a DUE.

It works fine on Repetier 1.0.3
[image]

With klipper on the other hand, the display outputs random data alsongside the correct temperatures.
Accessing a menue results in even more scrambled data. 
[image]

The Configured pins in Klipper are identical to the ones in Repetier.

https://github.com/repetier/Repetier-Firmware/blob/c09783a0f4f1bc2bedefc8df2c90709172ae74b2/src/ArduinoDUE/Repetier/DisplayList.h#L1235-L1239

```
[display]
lcd_type = st7920
cs_pin = PC19
sclk_pin = PC17
sid_pin = PC18
encoder_pins = ^PC13,^PB21
click_pin = ^PC15
```

Any idea, what's happening here and how this can be fixed?

Thanks, 
-Markus

[klippy.log](https://github.com/KevinOConnor/klipper/files/2569501/klippy.log)

KevinOConnor, 2018-11-11:
That's surprising.  If you look at the top of the file klippy/extras/display/st7920.py you'll see two variables - ST7920_CMD_DELAY and ST7920_SYNC_DELAY.  If you increase these variables and then run "sudo service klipper restart" after each change, does the problem go away?

-Kevin

voodoo-bravo, 2018-11-12:
Thanks for your Reply. 

I'm on a conference this week. I'll let you know if changing the delays worked out, when I am back at my printer.

-Markus

KevinOConnor, 2018-11-25:
If you pull the test code on the work-st7920-20181124 branch, and reflash the micro-controller, does it improve the results?

cd ~/klipper ; git fetch ; git checkout origin/work-st7920-20181124 ; sudo service klipper stop ; make ; make flash ; sudo service klipper start

voodoo-bravo, 2018-11-26:
With work-st7920-20181124 it seems a little less glitchy, but it's still not usable.
[image]

KevinOConnor, 2018-11-26:
That looks like it's the ST7920_CMD_DELAY being too low.  When you tried both delays set to .000072, did the display still glitch or did it not come up at all?  If the latter, try setting ST7920_SYNC_DELAY=.000072 and then try gradually increasing ST7920_CMD_DELAY (eg, try it at .000045).  Be sure to do a "sudo service klipper restart" after any code change.

-Kevin

voodoo-bravo, 2018-11-27:
This is on work-st7920-20181124 with both delays set to .000072
[image]

I tried ST7920_SYNC_DELAY=.000072 and gradually increasing ST7920_CMD_DELAY (starting from .000045, .000005 intervals). With both values on .000072 the glitching seems better but is still existent.

KevinOConnor, 2018-11-28:
That's really quite strange.  I haven't seen any reports from other users.  It definitely looks like a timing issue, but the only timeouts are the ones you've changed.  If you're familiar with C code, you could try increasing the "nops" in src/lcd_st7920.c .  (I've also pushed up another change to that branch with some more "nops" in a slightly different location.)

Are you powering the display with 5v or with 3.3v?

-Kevin

voodoo-bravo, 2018-11-28:
The Display itself is powered via 5v and all Data-Lines are level-shifted to 5v via the RADDS' onboard Levelshifter on the Display-Port. 

[image]

I'll test your changes this weekend and let you know how it went. 

-Markus

jakep82, 2018-11-28:
I'm guessing this is related to the level shifting.  I run this same type of screen on a Due, but without level shifting.  I have separate 5V connected to power the screen, and everything else is connected directly to pins on the Due.

KevinOConnor, 2018-11-30:
I agree with @jakep82 - that level shifter is a large unknown.  Each bit transmitted to the st7920 needs a certain hold time - it's not clear if the level shifter can match the transmit rate that Klipper generates.

I've run my st7920 display with 5V power and 3.3V signalling lines.  That is okay, as far as I know, because the signal lines are mcu->st7920 only.

-Kevin

voodoo-bravo, 2018-12-02:
As of now, I'm not able to connect the display to any other port on the RADDS as they are either blocked by something else, or would need more hardware modifications.

I'll stay with Repetier for now, also because of #879 

Thanks for your Time.

-Markus

closed 2018-12-02 by voodoo-bravo

voodoo-bravo, 2019-03-26:
Quick follow up on this issue, if anyone is facing the same problems. 
I had some time on hand to go over the electronics of my printer. I removed the levelshifter and bridged the contacts, so that the DUEs IO-Pins are directly connected to the RADDS Display-header. 

The display now works fine with Klipper. 

-Markus

kssdl888, 2021-09-21:
how to solve  this  problem ?