Klipper3d/klipper #885: Z probe

Kowy54, 2018-11-11:
How to setup Inductive sensor as Z endstop and probe?
Still fail.

Hywelmartin, 2018-11-11:
I think you must give us some more information....

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

rgrae81, 2018-11-12:
You have to define [probe] and [bed_mesh] in Printer-Config. If you use the Sensor on z-endstop pin then use the pin in section probe and under z-axis use: endstop_pin: probe:z_virtual_endstop

rgrae81, 2018-11-12:
Hi Kowy

I wrote it before (probe pin and endstop pin)
this should work if you use the sensor on the z-endstop-pin:

[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .0025
endstop_pin: probe: z_virtual_endstop
position_endstop: 0
position_max: 220

[probe]
pin: ^ar18
x_offset: 23
y_offset: 5
z_offset: -0.15

I think this is not a klipper issue.

Hywelmartin, 2018-11-12:
My config:
[stepper_z]
step_pin = ar46
dir_pin = ar48
enable_pin = !ar62
step_distance = .0025
endstop_pin: probe:z_virtual_endstop
#endstop_pin = ^ar18
position_endstop = 0
position_max = 220

[probe]
pin: ^ar15
x_offset: 23
y_offset: 5
z_offset: -0.15

I believe that the offset is wrong as well, it should be the distance between the nozzle and the bed when triggered. 

Else alter the min Z allowed

rgrae81, 2018-11-12:
> My config:
> [stepper_z]
> step_pin = ar46
> dir_pin = ar48
> enable_pin = !ar62
> step_distance = .0025
> endstop_pin: probe:z_virtual_endstop
> #endstop_pin = ^ar18
> position_endstop = 0
> position_max = 220 [...]

Same thoughts :) but you forgot to change the probe-pin to the originating endstop-pin

Kowy54, 2018-11-12:
I'll try it after work. Z_offset value I have from marlin.

rgrae81, 2018-11-12:
> 
> 
> I'll try it after work. Z_offset value I have from marlin.

It's written in the example-config. z_offset = distance between bed and nozzle when endstop is triggered. So if you use a negative value here, you say that when the endstop is triggered your nozzle is underneath your bed.

Your right offset should be positive in klipper: z_offset: 0.15

Kowy54, 2018-11-12:
Oh, OK. Understand

Hywelmartin, 2018-11-12:
@rgre81 I taught that the probe config was right....

Kowy54, 2018-11-12:
Error: Failed to home probe: Timeout during endstop homing after 6th probe point

Hywelmartin, 2018-11-12:
In printersection
[printer]
minimum_z_position:0.0

alter this to maybe -2.0 depending on how crocked your bed is

Kowy54, 2018-11-12:
Option 'minimum_z_position' is not valid in section 'printer'

Hywelmartin, 2018-11-12:
Oh is that a delta only thing...

Kowy54, 2018-11-12:
SO, printer is running. Some problems with model size, but its running :)

Hywelmartin, 2018-11-12:
You can try this under stepper z

position_min: 0

Alter to so e negativ number that see, s sane

KevinOConnor, 2018-11-23:
It looks like an answer was found for this issue.

closed 2018-11-23 by KevinOConnor

hfougere, 2021-07-02:
whats the config for BTT SKR 1.4 turbo? I'm trying to connect the induction level sensor that came with my tronxy x5sa pro