Klipper3d/klipper #886: Filament Tracking Sensor

AniInABox, 2018-11-12:
Hi,

would it be possible to get a Filament Tracking Sensor to work?
Something like this: https://sparklab-shop.de/elektronik/60/fts-v2-filament-tracking-system-1-75mm?c=10

As an Example: Repetier can slow down a print, when the Sensor is not sending the correct Pulses for the Extrusion Steps, so the print is saved.

Also an Extruder Jam can be recognized.

When you need some Debug Assistent, i can buy a Sensor and help you.

Best Regards

KevinOConnor, 2018-11-17:
I'm sure it is possible to add support, but it would require development work.  There's been some discussion on filament sensors in PR #545.

-Kevin

KevinOConnor, 2018-11-23:
I'm going to close this in favour of the existing issues on filament trackers (#249 and #545).

PR #3857 (2021-03-15): Add filament_motion_sensor
Add functionality to support a Filament Motion Sensor for detecting extruder jams as well as runouts.  Works by an encoder toggling the switch_pin 0/1 as the filament is pulled through the sensor.

Everything seems to work ok, except I have seen some weirdness in pause_resume.  When pausing, the printer runs the runout_gcode however it sometimes moves back over the print job (does not affect Z, only X (and maybe Y)).  Increasing the pause_delay may resolve this.

Also pause_resume seemed a little inconsistent in how it was handled so I tried to address, not sure if t [... cut]
closed 2018-11-23 by KevinOConnor