Klipper3d/klipper #889: rotate display under a parameter in the config file

manu7irl, 2018-11-13:
Hi everyone keep up this wonderfull work,
I am trying to install my lcd st7920  upside down'
and I am not able to find a simple way to do so in anyway.
Could someone point me to a place to start with?
this is the stock display for the Creality3D ENDER-3
also I am doing the review of the config file for it the config folder and it is not working,
I will post my changing and custom BLTOUCH config in a few minutes. 
thanks a lot!

`// Factory display for Creality CR-10
// https://www.aliexpress.com/item/Universal-LCD-12864-3D-Printer-Display-Screen-With-Encoder-For-CR-10-CR-7-Model/32833148327.html
//
// This is RAMPS-compatible using a single 10-pin connector.
// (For CR-10 owners who want to replace the Melzi Creality board but retain the display)
//
#define CR10_STOCKDISPLAY`

KevinOConnor, 2018-11-14:
Klipper does not support rotating the display.  Klipper uses the st7920 hardware character generator which does not (to the best of my knowledge) support screen rotation, so it would not be a simple change to add support.

-Kevin

manu7irl, 2018-11-14:
ok thank you so much, I will print a new case support rotated the the right way!
[](https://www.thingiverse.com/thing:3212875)
Thanks to Michael from the discord Klipper forum!

closed 2018-11-14 by manu7irl