Klipper3d/klipper #891: make menuconfig fails due to ncurses libraries missing - which cannot be installed ?!

fakio87, 2018-11-14:
Hi, 

I ran into the following on installing Klipper in the make menconfig process:

pi@octopi2:~ $ cd ~/klipper/
pi@octopi2:~/klipper $ make menuconfig
make: avr-gcc: Command not found
 *** Unable to find the ncurses libraries or the
 *** required header files.
 *** 'make menuconfig' requires the ncurses libraries.
 ***
 *** Install ncurses (ncurses-devel) and try again.
 ***
/home/pi/klipper/scripts/kconfig/Makefile:196: recipe for target 'scripts/kconfig/dochecklxdialog' failed
make[1]: *** [scripts/kconfig/dochecklxdialog] Error 1
Makefile:100: recipe for target 'menuconfig' failed
make: *** [menuconfig] Error 2
pi@octopi2:~/klipper $ cd ..
pi@octopi2:~ $ sudo apt-get install libncurses5-dev libncursesw5-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package libncurses5-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libncursesw5-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libncurses5-dev' has no installation candidate
E: Package 'libncursesw5-dev' has no installation candidate

Any idea or owrkaround how to solve this? It is a fresh installed followed by all available system upgrades.

Cheers
fakio

ndupont-net, 2018-11-14:
I have those ones on my instance 

```
pi@octopi-2:~ $ apt list --installed|grep ncur

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libncurses5/stable,now 6.0+20161126-1+deb9u2 armhf [installed]
libncurses5-dev/stable,now 6.0+20161126-1+deb9u2 armhf [installed]
libncursesw5/stable,now 6.0+20161126-1+deb9u2 armhf [installed]
ncurses-base/stable,now 6.0+20161126-1+deb9u2 all [installed]
ncurses-bin/stable,now 6.0+20161126-1+deb9u2 armhf [installed]
ncurses-term/stable,now 6.0+20161126-1+deb9u2 all [installed,automatic]

```

KevinOConnor, 2018-11-16:
Make sure you follow the directions described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md .  In particular, make sure you've run the `./klipper/scripts/install-octopi.sh` script.

-Kevin

peterus, 2018-11-22:
The first issue is already in this line:

> make: avr-gcc: Command not found

follow the installation guide, it will install the required packages.

fakio87, 2018-11-22:
Thanks, somehow it seems like the initial installation crashed midway and trying to execute the command again did not work till a restart. After the restart it worked flawlessly. Thank you for your help guys!

closed 2018-11-22 by fakio87

Afterschnitzel, 2020-11-13:
I have the same issue. I solved it by deleting klipper and clone it again.

`sudo rm - rf ~/klipper
cd ~/
git clone https://github.com/KevinOConnor/klipper
./klipper/scripts/install-octopi.sh`

Now it works.

redfox05, 2021-11-26:
I know this is old, but it's still coming up in search results. So anyone coming across this after getting the following errors:
-  During the klipper `make` command regarding avr-gcc or similar.
- Earlier on (less obviously) noticed an error during the `install-octopi.sh` script outputting something similar to `buster InRelease' changed its 'Suite' value from 'testing' to 'oldstable'`

The first error is likely due to missing packages which were not installed when the 2nd error was encountered during the `install-octopi.sh` script. It's not immediately obvious that this error about the repo is indeed an error, as one might assume its just a verbose info note.

A [solution I found](https://www.reddit.com/r/BIGTREETECH/comments/pr7h4c/error_when_installing_klipper_on_the_octopus/) for this is to run `sudo apt-get update --allow-releaseinfo-change` and then re-run the `install-octopi.sh` script. Or, edit the octopi script to add the release info change. Now warning tho, I think there are some potential side effects from doing this, as others have [mentioned here](https://forums.raspberrypi.com/viewtopic.php?t=318302). But those seem to be more about during a full upgrade?

Perhaps @KevinOConnor  might be able to explain better if this is a good solution or not, my linux knowledge is not that good with regards to OS updates.