Klipper3d/klipper #892: speed retraction

motorist828, 2018-11-16:
Hello!
I found that the clipper limits the speed of the retract, it is about 40 mm / s
I could not get more speed by increasing the values of max_extrude_only_ cycles, and max_extrude_only_axel does not work
Extruder motor works well without gaps
To test, I used this gcode and watched the extruder roller spin
After the value of 2600, I do not see the change in speed
Is this a clipper problem or is an arduino mega incapable of greater speed?
[testcode.txt](https://github.com/KevinOConnor/klipper/files/2587613/testcode.txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/2587578/klippy.log)

KevinOConnor, 2018-11-16:
The printer's overall max_accel, max_velocity, and max_accel_to_decel settings do also apply to extrude only moves.  The max_extrude_only_xxx settings can only reduce these acceleration/velocity limits.  In your test case, the overall limits are what is limiting the speed.

-Kevin

motorist828, 2018-11-16:
Strange why this was done that way?
Instead of a separate setting

KevinOConnor, 2018-11-23:
The printer settings currently apply to all movement (including extruder movement).  Perhaps it is not the best way, but it is what is currently implemented.

-Kevin

KevinOConnor, 2018-11-30:
It looks like the original question was answered, so I'm going to close this issue.

closed 2018-11-30 by KevinOConnor