Klipper3d/klipper #894: Keep getting: MCU 'mcu' shutdown: Stepper too far in past

Nandox7, 2018-11-16:
Every time I try to print this happens.
It homes all axes, extrudes a bit of filament as I have it set and then error.

\--
7:33:47.966: !! MCU 'mcu' shutdown: Stepper too far in past
17:33:47.966: // MCU 'mcu' shutdown: Stepper too far in past
17:33:47.966: // This generally occurs when the micro-controller has been
17:33:47.966: // requested to step at a rate higher than it is capable of
17:33:47.967: // obtaining.
17:33:47.967: // Once the underlying issue is corrected, use the
17:33:47.968: // "FIRMWARE_RESTART" command to reset the firmware, reload the
17:33:47.968: // config, and restart the host software.
17:33:47.968: // Printer is shutdown
17:33:47.968: Firmware was halted, trying to reconnect. Eventually running print is stopped.
17:33:48.979: !! MCU 'mcu' shutdown: Stepper too far in past
17:33:48.982: Response while unconnected:ok T0:46.0 /205.0 B:26.1 /40.0
\--

Hardware:
Radds 1.5 + Due
RAPS128 Drivers at 1/128 step
Raspberry Pi 3

This is the beginning of the GCode:
\--
G90
M82
M106 S0
M140 S40
M190 S40
M104 S205 T0
M109 S205 T0
; -- START GCODE --
;M190 S60
;M109 S205
G21                       	; metric values
G90                       	; absolute positioning
M107                     	; start with the fan off
;G28 X0 Y0		; X/Y/Z homing
G28 X0 Y0 Z0		; X/Y/Z homing
;G29		; auto-level process
;G32		; auto-level process
G1 X0 Y0 F6000		; move to X0 Y0 at 60mm/s
G1 Z15.0 F1200    	; move the platform down 15mm
G92 E0                  	; zero the extruded length
G1 F200 E15         	; extrude 10mm of feed stock
G92 E0                  	; zero the extruded length again
M117 Printing...		; put printing message on LCD screen
; -- end of START GCODE --
G92 E0
G1 E-2.6000 F2400
G1 Z0.700 F1200
; process Process1
; layer 1, Z = 0.200
T0
; tool H0.200 W0.470
; skirt
G1 X96.440 Y96.174 F7200
G1 Z0.200 F1200
\--

I suspect it may be failling at line
G1 E-2.6000 F2400
or 
G1 Z0.700 F1200

But don't see reason for it, I was running Repetier Firmware for it and those speeds always worked.

I did try to run the debug steps for ticks

[klippy.log](https://github.com/KevinOConnor/klipper/files/2590289/klippy.log)
[image]

jakep82, 2018-11-16:
The error says it all.  At your extruder step distance, you're running 5747 steps per mm.  At 40mm/s you're running about 230,000 steps per second.  Add in a Z move at 64,000 steps per second followed by an XY move at a combined 175,000 steps per second, and I think you're momentarily pushing about 469,000 steps.  You probably need to run your extruder at no more than 64 usteps, and even there you don't have a ton of overhead.

KevinOConnor, 2018-11-16:
It is failing on "G1 E-2.6000 F2400" - that's a request to move the filament at 40mm/s .  With an extruder step distance of .000174 that would result in ~230K steps per second, which is more than Klipper can step a single stepper (due to the need to pause 2us between each step and unstep event).

Double check the extruder step distance (it's rare to have an extruder with a nominal step distance measured in nanometers).  Otherwise, you can explicitly specify a lower max_extrude_only_velocity to limit extrude only speeds.

-Kevin

Nandox7, 2018-11-16:
Thank you both for the insights.
Probably Repetier FW imposes a max limit so that was not an issue, plus I think it would handle the missed steps if needed.

Well I have printed before and extruder step distance is correct, at least ended part (test cube) was within a very good tolerance and quality. No over extrusion detected.

finally those lines are added automatically by S3D, I'll review the profile settings.
The code I manually did and have direct control over end at "-- end of START GCODE --".

Question, is this something I can test using the Micro-controller Benchmarks steps?
I did try to go over that part is is basically impossible to do anything as it keeps outputting data in the console.
--
1536.183: analog_in_state oid=4 next_clock=2023951680 value=31702
1536.273: analog_in_state oid=13 next_clock=2027731680 value=31734
1536.482: analog_in_state oid=4 next_clock=2036551680 value=31700
1536.571: analog_in_state oid=13 next_clock=2040331680 value=31736
(...)
---
Using SUPPRESS stats or SUPPRESS analog_in_state,... doesn't stop it.

KevinOConnor, 2018-11-16:
To suppress the stats, you'd use:
```
SUPPRESS analog_in_state 4
SUPPRESS analog_in_state 13
```

Though, it's simpler to just reset the micro-controller and connect via console.py prior to klippy.py configuring the micro-controller.  Also possible by running `reset` via console.py (and then exiting and restarting console.py).

-Kevin

KevinOConnor, 2018-11-30:
It looks like the original question was answered, so I'm going to close this issue.

closed 2018-11-30 by KevinOConnor