Klipper3d/klipper #895: [Feature request] hot swap Klipper configs for different filaments.

TronskiFPV, 2018-11-16:
Maybe there is something more elegant, but I would like to see a way to select configs for different filaments/nozzles within the Klipper plugin for octopi to account mostly for pressure advance. I assume the ground work would need to happen with Klipper before it can be implemented in the plugin. 
Thanks for the great software. This project is really neat. 

Workaround:
change klipper config manually before using a different kind of filament.

jakep82, 2018-11-16:
This can already be accomplished with g-code macros.  Simply create separate macros that have the different pressure advance values that you want.  Pressure advance is run-time configurable, so you don't need to edit your config file.

TronskiFPV, 2018-11-16:
awesome thank you! I will try to add to the docs when I try this.

npgoalkeeper, 2018-11-17:
Slic3r PE has support for putting in the macro, and you don't need to guess where the filament will change.

TronskiFPV, 2018-11-21:
Thanks! I'm still working on making the macro...
for now I'm just commenting out the values of the filament I'm not using in the config file.

KevinOConnor, 2018-11-30:
It looks like the original question was answered (use a gcode_macro or similar run-time setting), so I'm going to close this issue.

closed 2018-11-30 by KevinOConnor