Klipper3d/klipper #896: Position signals instead of step/direction signals?

npgoalkeeper, 2018-11-16:
Is there any way that a raspberry pi running klipper provide position signals instead of step/dir signals in order to use external cDriver and motion controller IC's for future proofing (IC's such as the TMC4361, 5161, and 4671)? I want to us e FOC and servo control on my 3D printer to reach extreme speeds in excess of 300mm/sec, and a standard stepper motor will not be able to provide me with the amount of torque or speed I require (the motors are geared, 5:1 on all axis, 40:1 on the extruder).

michaelwu21, 2018-11-17:
^
Would be nice if klipper had an option to sends step/dir through spi. Doesn't seem to complicated as it just needs to write the target position(in steps) to a register.

KevinOConnor, 2018-11-17:
It may be possible to control a motor via UART/SPI, but I suspect it would be difficult.  All the motion controllers I've seen can accept a step/dir input, so it seems simpler to continue to use that method.  Klipper can create more than enough step/dir pulses so there's no real downside to using it.

Note that it is not sufficient to send a position to a motion controller as motor movement needs to be tightly synchronized with other motors.  What's needed is to send a series of fine grained "positions" and "time at each position".  The step/dir interface is a compact way of doing just that.

-Kevin

npgoalkeeper, 2018-11-17:
Though the TMC4361 I'm using can use step/dir, it bypasses the integrated ramping generator, rendering the TMC4361 into what is seemingly an expensive TMC5130 when used in conjunction with a TMC2224. If synchronizing speed is such an issue, then dcStep technology can be deployed in order to synchronize the speed that each motor runs at. This just needs to be set-up to run in a TRAMS setup, meaning that this has already been integrated in Marlin, so you can use that for reference if you do decide that this is a feature that you wish to integrate in the future.

npgoalkeeper, 2018-11-17:
Just to ensure that you understand my suggestion, the position signal is just a position for the motor to go to at a set acceleration/deceleration, and the IC itself generates the ramping, which is s-shaped. In other words, the step/dir signals are generated by the IC rather than modified by the IC. I understand that you already have an existing s-shaped ramping version of klipper, however it is not merged with the main branch and is therefore not updated that often.

npgoalkeeper, 2018-11-17:
I could already get closed loop operation via another microprocessor. What I can't get is hardware FOC and sixpoint. It seems like sixpoint could be more effective than s shaped ramping for 3D printing.

cmock, 2018-11-19:
The ESP8266 has to do regular calls to the WiFi firmware, so I don't think it is real-time capable in the sense that we need here.

npgoalkeeper, 2018-11-19:
I thought the ESP8266 could be used without wifi. also, when I asked Trinamic support about the TMC4671 using the step/dir interface, they responded with:

 Hi, 

the TMC4671 Step/Direction Input does not work correctly (please see errata of TMC4671 for further information). 
You can use SPI/UART/PWM/Analog Interface to control the motor. We will fix this with a new version. 

Hall Effect current sensors willl work well and makes sense for higher current application. You can use them instead the current resistor. 

Best regards from Hamburg
Joshua Pham Xuan

KevinOConnor, 2018-11-20:
Klipper generates movements based on the mathematical formulas for the printer's kinematics and it calculates very precise timing from that.  It's not limited to constant acceleration ramps (nor even s-curve ramps).  In my brief look at various motion controller specs, I did not see a way to maintain that type of precision when using uart/spi/i2c type interfaces.  To wit, Klipper doesn't want to "move from position 100 to position 200 at 200 steps per second" - instead Klipper wants to "move from position 100 at time 12:27:34.576213 to position 200 at 200 steps per second and arrive at time 12:27:35.076213".  The step/dir interface does allow Klipper to maintain that type of precision.

That said, if someone wishes to do the development for alternative interfaces, then that is fine.

-Kevin

closed 2018-11-20 by npgoalkeeper