Klipper3d/klipper #897: [Feature request] kinematics class for n axis CNC lab equipment

D1plo1d, 2018-11-16:
Klipper's realtime stepper control has applications beyond the scope of 3D printers as generalized CNC control.

I'm looking into using Klipper to drive up to 10 axes in parallel. Each axis is actuating a syringe in a CNC microfluidics setup. Am I right in thinking that I would only need to create a new kinematics class based on `CartKinematics` with a rails definition that supports `stepper_aa` to `stepper_zz` to support up to 26*26 = 676 axes or are there other limitations?

KevinOConnor, 2018-11-17:
If you're just looking to drive multiple steppers in parallel then no code change is needed.  Klipper already supports parallel steppers - see the "[stepper_z1]" section in the config/example-extras.cfg file for the details.  There is no limit on the number of parallel steppers.

If you're looking to define custom robot kinematics, then yes - you could create a new kinematics class and define any number of steppers (with whatever names you wish).  There's some details on the process of creating new kinematics at: https://github.com/KevinOConnor/klipper/blob/master/docs/Code_Overview.md#adding-new-kinematics

Cheers,
-Kevin

D1plo1d, 2018-11-23:
Thanks Kevin. z1 would work for my purpose if there was a way to move each z axis independently at the same time with kinematics similar to how `G1` commands move x and y independently at the same time.  I've looked at `Z_TILT_ADJUST` and `FORCE_MOVE` but neither of them appear like they'd be able to achieve movement on multiple z steppers to arbitrary positions without breaking the kinematics and needing a `G28`.

If I am right in thinking that there is no way to do a coordinated move of more then 3 independent axes at the same time with Klipper as it stands today then it we may wish to add a Extended GCode which overcomes the limitations of `G1` which I would call `MOVE` to move an arbitrary number of axes with no restriction on names so an axis name may include numbers such as `Z1` or be named `F` which would currently collide with Feedrate in a G1.

Strawman example: `MOVE { "axes": { "x": 10, "y": 20, "z1": 30, "z2": 40, "z3": 50}, "feedrate": 20}`

Thanks again,
Rob

D1plo1d, 2018-11-23:
Just had another thought, `MOVE_TO` and `MOVE_BY` might be semantically nicer for developers for absolute and relative movement respectively. This follows the macro names I've established for the secure 3D printer network host I'm developing ( [Tegh](https://github.com/tegh/tegh/tree/master/packages/tegh-macros-default/src/macros) ). Developing a UI against Tegh macros I've found MOVE_TO and MOVE_BY to have nice API ergonomics in practice.

D1plo1d, 2018-11-23:
Hmm, if this is generalised movement making no assumptions about the machine's axis configuration then feedrate is ambiguous, begs questions "is that feedrate number the velocity vector produced by geometrically combining all the axes in the movement? Are Extruder axes and Z special-case?".

Strawman 2: coordinated movement for CNC lab equipment that has X, Y, Z2, Z3 and E1 all able to move at 20mm/s but a Z1 axis that can only move at 5mm/s (E1 is contrived here because I would just use another Z axis but it demonstrates the flexibility of the approach):
```
MOVE {
  "axes": { "x": 10, "y": 20, "z1": 3, "z2": 40, "z3": 50, "e1": 60},
  "feedrate": [
    {"max": 20, axes: ["x", "y", "z2", "z3", "e1"]},
    {"max": 5, axes: ["z1"]}
  ]
}
```

KevinOConnor, 2018-11-25:
Unfortunately, I don't understand what robot motion you are trying to accomplish, so there's little advice I can provide.

In Klipper, the motion planner takes a starting XYZ coordinate and ending XYZ coordinate (all in cartesian space) and the kinematics code schedules the necessary stepper movements to arrange for the toolhead to travel along that line of movement.  It's not clear to me what a "Z2" coordinate is.

If you're not interested in motion planning of a toolhead and instead just wish to move a set of steppers an arbitrary distance simultaneously, then you don't need to use the kinematics code.  An example of this would be the klippy/extras/force_move.py:manual_move() code.  That method itself will only move one stepper at a time, but one could write a similar function that schedules the simultaneous movement of multiple steppers if one wished.

-Kevin

KevinOConnor, 2018-12-17:
Any further comments on this?  If not, I'm inclined to close this issue.

-Kevin

D1plo1d, 2018-12-18:
Yeah, that's fine. I'm going to work on a n-axis rewrite of the host software as a plugin to  https://github.com/tegh/tegh

closed 2018-12-18 by D1plo1d

KevinOConnor, 2019-03-03:
FYI, Klipper now supports defining "manual steppers" which can be independently controlled.  (Details in config/example-extras.cfg.)

-Kevin

D1plo1d, 2019-03-04:
I will definitely check that out. Thanks Kevin!

D1plo1d, 2019-03-13:
@KevinOConnor Thanks again for this feature. We got `MANUAL_STEPPER` working for individual stepper control however for our microfluidics to work we need to drive 2 or more steppers in coordination (one or more to drive reagent/fluid into the microfluidic chip and one to remove the same amount of water from the chip). As a workaround I am going to generate 1 `MANUAL_STEPPER` gcode per step and repeatedly scan through the motors (eg. motor A moves 1 step then motor B moves 1 step then motor C moves 1 step, repeat) but this is not ideal either from a gcode readability, file size or efficiency perspective. It should at least get the microfluidic syringes moving synchronously (albeit potentially slowly due to the flood of gcodes).

The ability to do a coordinated movement of multiple manual steppers however would be ideal.

KevinOConnor, 2019-03-18:
Okay, thanks for the feedback.  I'm not sure what type of coordination you need, but if it as simple as running multiple moves at the same time, then I think that's pretty straight forward to implement.  It just involves disabling some of the time synchronization.

As an example (totally untested) something like:
```diff
--- a/klippy/extras/manual_stepper.py
+++ b/klippy/extras/manual_stepper.py
@@ -45,7 +45,6 @@ class ManualStepper:
     def do_enable(self, enable):
         self.sync_print_time()
         self.stepper.motor_enable(self.next_cmd_time, enable)
-        self.sync_print_time()
     def do_set_position(self, setpos):
         self.stepper.set_position([setpos, 0., 0.])
     def do_move(self, movepos, speed, accel):
@@ -60,7 +59,6 @@ class ManualStepper:
                        0., cruise_v, accel)
         self.stepper.step_itersolve(self.cmove)
         self.next_cmd_time += accel_t + cruise_t + accel_t
-        self.sync_print_time()
     def do_homing_move(self, movepos, speed, accel, triggered):
         if not self.can_home:
             raise self.gcode.error("No endstop for this manual stepper")
@@ -92,8 +90,8 @@ class ManualStepper:
             except homing.EndstopError as e:
                 if error is None:
                     error = str(e)
-        self.sync_print_time()
         if error is not None:
+            self.sync_print_time()
             raise self.gcode.error(error)
     cmd_MANUAL_STEPPER_help = "Command a manually configured stepper"
     def cmd_MANUAL_STEPPER(self, params):
@@ -115,6 +113,8 @@ class ManualStepper:
             if 'ENABLE' not in params and not self.stepper.is_motor_enabled():
                 self.do_enable(True)
             self.do_move(movepos, speed, accel)
+        if self.gcode.get_int('SYNC', params, 1):
+            self.sync_print_time()
     def handle_motor_off(self, print_time):
         self.do_enable(0)
 
```

With this, one would use `MANUAL_STEPPER SYNC=0 ...` commands to take an action without waiting for the move to complete.

-Kevin

D1plo1d, 2019-03-20:
That's great! SYNC looks like it will solve our remaining issues. To drive all 10 syringes I expect I could run 9 `MANUAL_STEPPER  SYNC=0` G Codes followed by one `MANUAL_STEPPER` G Code so that the final stepper's move blocks until the synchronised multi-stepper move is complete.

KevinOConnor, 2019-03-20:
Right.  (If it's possible for the 10 steppers to require different amount of times, then run 10 with "manual_stepper sync=0 move=123" followed by 10 with "manual_stepper sync=1".  Let me know if you can make the change above and test it - otherwise I can add it to the code in a few days.

-Kevin

D1plo1d, 2019-03-22:
Makes sense. I'm not going to have a chance to work on this in the next week but it would be awesome if you can merged it in.

KevinOConnor, 2020-03-25:
FYI, basic support for moving manual_stepper motors in parallel was added as part of commit 6be552f.

-Kevin

Nickperillo, 2021-06-08:
Using SYNC=0 doesn't seem to work at all. 

Here is a macro I'm trying to implement:
```
[gcode_macro STAND_UP]
gcode:
  MANUAL_STEPPER STEPPER=leg_left MOVE=120 SYNC=0
  MANUAL_STEPPER STEPPER=leg_right MOVE=120 SYNC=0
  MANUAL_STEPPER STEPPER=leg_back MOVE=120 SYNC=0
```

When I try this, the "legs" in my case move one after the other in sequence, rather than simultaneously. 

Is there something I'm missing? 

I'm trying to use Klipper to prototype a basic robotic mechanism using gcode macros.