Klipper3d/klipper #898: [Feature request] Anet A8 Buttons

therealpanse, 2018-11-17:
It was mentioned once here, that the buttons on an Anet A8 don't work with Klipper. Could there be at least some functionality (eg. Emergency Stop, Pause, etc.) with those buttons? I hate to see, that something went wrong and having to run to my pc to stop the print or just janking the plug...

I mean, if that works I wouldn't say no to a fully implemented menu of any kind, but for safety reasons it would be nice to stop the print without shutting the printer down completely.

KevinOConnor, 2018-11-18:
It's certainly possible to support it, but it requires someone to do the necessary software development work.  There's some details in issue #609.

-Kevin

mcmatrix, 2018-11-21:
@therealpanse Does your anet has this display and keypad
[image]
In theory it should be possible to read that ADC pin periodically (is 100ms enought for button?, like thermistors) and monitor readout change. When readout change is noticeable then call registred callback with value.

It requires change in `buttons` module to support adc change callbacks and change in `menu` module to
call right button depending from adc value. It would be nice when concrete button adc range(min,max) can be inserted into config. For example
`click_pin: ar37(300,400) # if the adc range is from 0 to 1023`

If nobody is working on this then I can try to make inital support for such functionality.

therealpanse, 2018-11-25:
Sorry for answering that late. Yes, that's the one. Haven't heard of anyone working on it right now.

derbengale, 2018-11-26:
According to this schematic [here](https://github.com/ralf-e/ANET-3D-Board-V1.0/blob/master/ANET3D_LCD2004_board.pdf
) the keys should be connected to analog pin A1. I just tried to read out the analog values with this Arduino sketch: `void setup() {
  // put your setup code here, to run once:
Serial.begin(115200);
}

void loop() {
  int A0=analogRead(A0);
  int A1=analogRead(A1);
  int A2=analogRead(A2);
  int A3=analogRead(A3);
  int A4=analogRead(A4);
  int A5=analogRead(A5);
  int A6=analogRead(A6);
  Serial.print(A0);
  Serial.print(" ");
  Serial.print(A1);
  Serial.print(" ");
  Serial.print(A2);
  Serial.print(" ");
  Serial.print(A3);
  Serial.print(" ");
  Serial.print(A4);
  Serial.print(" ");
  Serial.print(A5);
  Serial.print(" ");
  Serial.println(A6);
 
}` unfortunately this didnt work out as I expected. All the analog readouts give me the same values. When i restart the serial plotter of the Arduino IDE they settle at different values even. And they all seem to react in the same way when I push any buttons. Am I doing something wrong?
[image]

mcmatrix, 2018-11-26:
@derbengale Thanks for the schematics. On what board are you trying to connecting that display?

derbengale, 2018-11-26:
I am using the Anet 1.5 Board but there are no schematics for that one I guess. I think nothing has changed between the revision since we are also using board definitions for the Anet 1.0 board. Do you have any idea why all the analog inputs were showing the same value? I have some experience to arduino stuff but never experienced something like this. Is the pin descriptor wrong?

derbengale, 2018-11-28:
Is there noone who can help us Anet users? I would like to do it, but I dont know where to start with the python host. I never used Python before. Is there a minimal host (just the serial communication with the Klipper host) that would be a good start for me. I think it will get very complicated for me to understand where and how I can merge it into the Klipper client software. I would like to help though but I guess it wouldnt be useful for you guys if I programmed anything in C right?

ZhenyaKa, 2018-11-30:
> click_pin: ar37(300,400) # if the ADC range is from 0 to 1023 up_pin: ar37(450,550) down_pin: ar37(600,700)

Here values from ui.h for zonestar five buttons.

void uiCheckKeys(uint16_t &action) {
	struct {
		uint16_t min;
		uint16_t max;
		uint16_t action;
		} keys[] = {
		{   300,   500, UI_ACTION_BACK           },    // Left
		{   570,   870, UI_ACTION_PREVIOUS       },    // Up
		{  1150,  1450, ADC_KEYPAD_CENTER_ACTION },    // Center
		{  1900,  2200, UI_ACTION_OK             },    // Right
		{  2670,  2870, UI_ACTION_NEXT           }     // Down
	};

derbengale, 2018-12-19:
Thanks a lot. I was really glad to see some progress going on here so I just installed your branch of Klipper and checked if the changes you made to buttons.py are in my files. I then compiled it and used the menu configuration of  @rmcbc which was posted [in here](https://github.com/KevinOConnor/klipper/pull/977). Then I changed the pin decriptors to match my the Anet boards pin decriptors as following `[display]
lcd_type: hd44780
rs_pin: PA3
e_pin: PA2
d4_pin: PD2
d5_pin: PD3
d6_pin: PC0
d7_pin: PC1
click_pin: PA1(490,530)   # 510
back_pin: PA1(72,112)     # 92
up_pin: PA1(675,715)      # 695
down_pin: PA1(159,199)    # 179`
Unfortunately the buttons dont work yet.
The display works but when I press any of the buttons nothing happens. I attached my printer.cfg.
Does someone have any idea how to get it working?
[printer.cfg.zip](https://github.com/KevinOConnor/klipper/files/2695713/printer.cfg.zip)

ZhenyaKa, 2018-12-19:
> > click_pin: PA1(490,530)   # 510
> > back_pin: PA1(72,112)     # 92
> > up_pin: PA1(675,715)      # 695
> > down_pin: PA1(159,199)    # 179
> 
> Unfortunately the buttons dont work yet.

Ranges reading not realized yet in Klipper as I know

rmcbc, 2018-12-19:
You have to press 3 or 4 seconds the CLICK button for open de menu, (right key o ANET LCD).

mcmatrix, 2018-12-19:
> The longpress seems to need that the click button be pressed 3 or 4 seconds them release and after that it goes to menu.

With a normal default screen, a short click is needed to enter the default menu.

Custom screen (deck) only needs long-press to enter the additional menu. 
Long press has to be longer than 1s. Currently you have to also release the button.
`press&hold>1s......release`

mcmatrix, 2018-12-19:
@derbengale Try to set `adc_debug: True` attrib in `display` section.
It will continiously output ADC reading to the log file.
Then in putty (ssh) console use command
`tail -f /tmp/klippy.log`
It will give you live stream of log.
`ctrl+c` will exit from log
Then by pressing buttons and looking button ADC value.
You can build your range, min and max (adc_value-20, adc_value+20).

derbengale, 2018-12-19:
It was my bad: i just didnt  press the click pin long enough. I didn't know that. It was working all the time. :+1:  hooray :-)

mcmatrix, 2018-12-19:
> It was my bad: i just didnt press the click pin long enough. I didn't know that. It was working all the time.

@derbengale Just a small footnote in case  you blindly copied config.
If you don't have special need then there's no point of overriding all default menu
items in your printer config. 
If you want to change just one submenu then just override that concrete one.
By overriding all the default items you are cut away from klippers default menu future changes.

KevinOConnor, 2019-02-08:
Lets close this issue and track it in PR #977.

-Kevin

closed 2019-02-08 by KevinOConnor