Klipper3d/klipper #899: [Feature Request] (Menu) Interactive status screen + Shortcuts

NordicGuy, 2018-11-17:
@mcmatrix 

For some time I’ve had a couple of ideas that could potentially ease the users interaction with the LCD menu system. 

First; Interactive status screen
Currently the underlying menus are reached from the status screen via a click of the encoder. But what if, when rotating the encoder in a certain direction, fx right, you’d be able to access all the elements on the status screen directly ? The action would introduce  a curser to the status screen, fx an arrow or simply highlight the elements, that would toggle between the screens elements by rotating and if the user clicked on one would bring the user directly to its adjustment menu. 
This way you could quickly access any setting quickly and with minimum navigation from the status screen. 

Second; Menu shortcuts
This follows in a bit of the same spirit. That the user could assign any menu item in the config with a specific shortcut action; this could be a long press, double click, rotating the encoder while holding it down, any fast and “unusual” interaction with the encoder really. Fx on the default MK3 firmware, and possibly others, a long press of the encoder takes you straight to the move Z menu, which is quite practical in some situations. But it would be great to be able to chose this at will.

Thank you

mcmatrix, 2018-11-20:
@NordicGuy

Thanks for your feedback.
Actually, the `Interactive status screen` is already there.
You could examine `menu.cfg`
There are some predefined deck elements.
```
__screen2004_static
__screen2004_input
__screen1604_static
__screen1604_input
__screen_test
```
The deck contains cards and card has menu items definitions.
Check `example-menu.cfg` for info about menu items.

You can use these decks by setting deck name to menu_root attribute
```
[display]
menu_root: __screen2004_static
```

KevinOConnor, 2018-11-30:
It looks like the original question was answered (use a custom set of "card" menus), so I'm going to close this issue.

closed 2018-11-30 by KevinOConnor