Klipper3d/klipper #901: can't pass make command

Houba28, 2018-11-18:
I am following the installation.md, and everything works fine, unless I try to do the make command (Building and flashing the micro-controller) 
pi@octopi:~/klipper $ make V=1
mkdir -p out//scripts/kconfig/lxdialog
mkdir -p out//include/config
make -C out/ -f /home/pi/klipper/scripts/kconfig/Makefile srctree=/home/pi/klipper src=scripts/kconfig obj=scripts/kconfig Q= Kconfig=/home/pi/klipper/src/Kconfig  silentoldconfig
make[1]: Entering directory '/home/pi/klipper/out'
cc -Iscripts/kconfig -I/home/pi/klipper/scripts/kconfig   -DCURSES_LOC="<ncurses.h>" -DLOCALE  -c -o scripts/kconfig/conf.o /home/pi/klipper/scripts/kconfig/conf.c
cat /home/pi/klipper/scripts/kconfig/zconf.tab.c_shipped > scripts/kconfig/zconf.tab.c
cat /home/pi/klipper/scripts/kconfig/zconf.lex.c_shipped > scripts/kconfig/zconf.lex.c
cat /home/pi/klipper/scripts/kconfig/zconf.hash.c_shipped > scripts/kconfig/zconf.hash.c
cc -Iscripts/kconfig -I/home/pi/klipper/scripts/kconfig  -Iscripts/kconfig -DCURSES_LOC="<ncurses.h>" -DLOCALE  -c -o scripts/kconfig/zconf.tab.o scripts/kconfig/zconf.tab.c
cc  -o scripts/kconfig/conf scripts/kconfig/conf.o scripts/kconfig/zconf.tab.o
  Build Kconfig config file
mkdir -p include/config include/generated
scripts/kconfig/conf --silentoldconfig /home/pi/klipper/src/Kconfig
make[1]: Leaving directory '/home/pi/klipper/out'
  Creating symbolic link out/board
mkdir -p out/src out/src/avr out/src/generic
touch out/board-link
ln -Tsf /home/pi/klipper/src/"avr" out/board
mkdir -p out/board-generic
ln -Tsf /home/pi/klipper/src/generic out/board-generic/board
  Compiling out/src/sched.o
avr-gcc -Iout/ -Isrc -Iout/board-generic/ -std=gnu11 -O2 -MD -g -Wall -Wold-style-definition -Wtype-limits -ffunction-sections -fdata-sections -flto -fwhole-program -fno-use-linker-plugin -mmcu="atmega2560" -c src/sched.c -o out/src/sched.o
Assembler messages:
Error: unrecognized option -mmcu=avr6
Makefile:64: recipe for target 'out/src/sched.o' failed
make: *** [out/src/sched.o] Error 1

Also, log file content:
pi@octopi:~/klipper $ cat /tmp/klippy.log
=============== Log rollover at Sun Nov 18 15:52:34 2018 ===============
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.6.0-611-gbb41765'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Nov 24 2017, 17:33:09) \n[GCC 6.3.0 20170516]'
Building C code module c_helper.so
Start printer at Sun Nov 18 15:53:23 2018 (1542556403.7 1031.2)
Unable to open config file /home/pi/printer.cfg
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/configfile.py", line 96, in _read_config_file
    f = open(filename, 'rb')
IOError: [Errno 2] No such file or directory: '/home/pi/printer.cfg'
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 136, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 119, in _read_config
    config = pconfig.read_main_config()
  File "/home/pi/klipper/klippy/configfile.py", line 173, in read_main_config
    data = self._read_config_file(filename)
  File "/home/pi/klipper/klippy/configfile.py", line 102, in _read_config_file
    raise error(msg)
Error: Unable to open config file /home/pi/printer.cfg

I know the log says its missing some config for a printer, but from installation.md I think it will be created later, and it should not be the problem...
I checked avr-gcc avr-lib are present, only thing i do not understand is the flag -mmcu where by verbose mod is put an "atmega2560" but the error says "avr6"
how to fix this? I have MKSgen 1.4, so I chosed same as for arduino mega...

KevinOConnor, 2018-11-20:
The above looks like there is something wrong with the Octopi image.  You could try rerunning `./scripts/install-octopi.sh` and then try running `cd klipper ; make distclean ; make menuconfig ; make`.  If that doesn't work, try building the Octopi image again from scratch.

-Kevin

Houba28, 2018-11-20:
There was really something with my octopi image. I reinstalled whole Pi yesterday and everything is OK :) thank you

closed 2018-11-20 by Houba28