Klipper3d/klipper #902: Z-probe on standard kossel not triggering:

KeesBlokland, 2018-11-18:
I see Kevin is close to releasing a new version, if all my questions will be solved by that release, I am happy to wait!

My printer is a Kossel Delta with a Ramps 1.4 and Arduino mega (2560)
This is a 4 year old working system, with Z-probe. 
Z- Probe pin is Z-, which is actually Port D3, pin 46 on the mega. It is also known as logical pin 18.
The Z+ endstop is normally connected to logical pin 19. Z+ is a homing switch. more on that later. 

The doc states:
_Basic delta calibration

_Klipper has a DELTA_CALIBRATE command that can perform basic delta calibration. .....
There are two ways to perform the probing - manual probing and automatic probing. Automatic probing utilizes a hardware device capable of triggering when the toolhead is at a set distance from the bed. 
Manual probing involves using the "paper test" to determine the height at each probe point. 
It is recommended to use manual probing for delta calibration. A number of common printer kits come with probes that are not sufficiently accurate (specifically, small differences in arm length can cause effector tilt which can skew an automatic probe). Manual probing only takes a few minutes and it eliminates error introduced by the probe.
To perform the basic probe, make sure the config has a [delta_calibrate] section defined and run:
G28
DELTA_CALIBRATE METHOD=manual
After probing the seven points new delta parameters will be calculated.  Save and apply these parameters by running:_
SAVE_CONFIG_

In order to make something Probe, we need a 'probe' in printer.cfg. 
However, and before we get into that: 

After the G28 en issuing Delta_Calibrate, what is supposed to happen?
Is there something I need to do beforehand? 
 
I Send: G28Recv: ok[...]
Send: DELTA_CALIBRATE METHOD=manualRecv: ok

The head moves down to: 
Send: M114Recv: X:0.000 Y:0.000 Z:10.000 E:0.000

and that's all she wrote. Nothing else happens, no messages. 

Query_probe
Recv: // probe: open

Manually toggle state of probe: still open, it does not see the change on the pin.

Now for an experiment: Z+ is the pin (19 in arduino speak) that is normally my rear tower end switch. 

When I swap my Z+ and Z- (tower endswitch an probe switch) and make the proper changes in printer.cfg and hardware, I expect to be able to make 'Query_Probe' see a change when I trigger my known to be good switch.

Swapping the 2 switches in printer.cfg (and in real life) does not make a difference, but in both instances the homing works as designed. Which means, the switches are good, and being detected. 

The bit about the virtual Z-probe.  This is my mystery bit.

Why do I need a virtual pin, when I do have 2 distinct hardware pins. 
I can imagine this to be necessary when you only have one pin connected to a single arduino port/pin. Not when you have 2 inputs?

Why not do as Kevin says?  But, my printer.cfg has a stepper_C, not a stepper_Z. 
Should it be a c_virtual_endstop then?  (no is the answer, unless you rewrite probe.py)

[probe]
D3 is portname, logical pin 18
pin: ar18

My code:

The stepper_c section describes the stepper controlling the rear tower (at 90 degrees).
[stepper_c]
step_pin: ar46
dir_pin: !ar48
enable_pin: !ar62
step_distance: .01
#endstop_pin: ^ar19
endstop_pin: probe:z_virtual_endstop
#position_endstop: 161.42

Kevin's Example: 

[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
position_max: 200

If I use the virtual switch I loose the homing switch, and motor for C tries to escape.

So, how do I define (or where do I define) the probe:z_virtual_endstop??
Answer: It is all done in probe.py (I think).

but here's the problem: _In class PrinterProbe:_
...
    if config.has_section('stepper_z'): 

It is actually scanning the printer.cfg for a stepper_z, and it will not find one in my printer.cfg. A looked at a few examples, most seem to have the notation of stepper_a
When I boldly replace stepper_a with stepper_x, I get a complaint: no stepper_a.
So all that is wired up elsewhere I guess. 

>> Is that the problem? (I do not have a lot of python knowledge (yet)

But then again, I have 2 different input's why the need for virtualisation?

(yeah, I know, RTFM and you will get there ;-)

[klippy.log](https://github.com/KevinOConnor/klipper/files/2593106/klippy.log)

BlackStump, 2018-11-18:
First up you need to remember that the docs are somewhat generic in nature and virtual endstops are generally not needed for Deltas as they home to max. but are needed for Cartesian printer where the probe is also used as a endstop.
I dont see any standout issues in your config maybe try ^ar18 for your probe pin

https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md

The manual method of delta calibrate does not need a probe ie 
 Manual probing involves using the "paper test" to determine the height at each probe point.
jog down to the paper thickness then type  "NEXT" is my understanding tho I have not used that method.

KeesBlokland, 2018-11-19:
Thank you. 
I have tried negating the input, sorry to have omitted that, but maybe I should try a pullup resistor. The pin might not have one enabled. (for an old hardware engineer that is the fastest way to try).

I have not tried simply typing Next in the terminal. Again confused by the section under Klipper, where you enter coordinates and also push next.

KeesBlokland, 2018-11-19:
..at least it is clear I should not worry about the z probe at the moment.

KeesBlokland, 2018-11-19:
It's all clear now: All commands are entered in the terminal. Do not use anything else!

BlackStump's comment: _jog down to the paper thickness then type "NEXT" is my understanding_  was what I needed. Thanks for the trigger!

After the initial **G28** and **DELTA_CALIBRATE METHOD=manual** the probe moves to the first point. 
Then you use the printer panel to move the head down. (Under _Control_, go down to _Move 1_ (or 10 or 0.1) and select _Move Z_. 
The head moves down each time you select a value and push the knob. Once you reach the magic point where a piece of paper just slides under the head, go back to the terminal and type _NEXT_. and the head moves to the next point. After 7 points  you will see the following in the terminal window:

Send: NEXT
Send: NEXT
Recv: // stepper_a: position_endstop: 163.414872 angle: 209.884718 arm: 335.000000
Recv: // stepper_b: position_endstop: 162.603671 angle: 329.344298 arm: 335.000000
Recv: // stepper_c: position_endstop: 162.401223 angle: 90.000000 arm: 335.000000
Recv: // delta_radius: 175.332221
Recv: // The SAVE_CONFIG command will update the printer config file
Recv: // with these parameters and restart the printer.Recv: ok[...]Send: SAVE_CONFIG

... and now it all makes sense ;-) 
Sometimes a man does not see the obvious in front of his nose.

closed 2018-11-19 by KeesBlokland