Klipper3d/klipper #906: Disable specific stepper

cfelicio, 2018-11-19:
I'm trying to setup a filament runout sensor via Octoprint, and everything works great, except when it's time to put in the new filament, things would be a lot easier if I could manually move the extruder (it's a D-Bot, with a long bowden tube).

I tried disabling just the extruder for this purpose (M18 E), but Klipper disables all steppers. Is this the correct behavior? Any workarounds?

By disabling all the steppers, I'm not able to restart the print, as I would have to re-home all axis, losing my existing Z position.

Thanks! :-)

bruce356, 2018-11-19:
Hi, as soon as you disable a stepper and move the extruder manually you have lost your X-Y location and need to home again before continuing  the print. Why not pause your print and send the extruder to a pre-determined location, change your filament and resume/continue the print. The extruder will return to its correct location and continue the print.

In Octoprint Under GCODE Scripts
Under Heading "After print job is paused"

```
{% if pause_position.x is not none %}
; relative XYZE
G91
M83

; retract filament of 0.8 mm up, move Z slightly upwards and 
G1 Z+5 E-2.0 F4500

; lower tool temperature if needed
M104 S105

; absolute XYZE
M82
G90

; move to a safe rest position, adjust as necessary
G1 X-40 Y0
{% endif %}

```

Under Heading "Before print job is resumed"

```
{% if pause_position.x is not none %}
; relative extruder
M83

M104 S210 T0; set extruder temperature
M109 S210 T0; Wait for Hotend Temperature

; prime nozzle
G1 E-0.8 F4500
G1 E2.8 F4500
; G1 E0.8 F4500

; absolute E
M82

; absolute XYZ
G90

; reset E
G92 E{{ pause_position.e }}

; move back to pause position XYZ
G1 X{{ pause_position.x }} Y{{ pause_position.y }} Z{{ pause_position.z }} F4500

; reset to feed rate before pause if available
{% if pause_position.f is not none %}G1 F{{ pause_position.f }}{% endif %}
{% endif %}

```
You would need to modify to suit your printer.
regards - bruce

cfelicio, 2018-11-19:
Hi Bruce,

That's the script I'm using to pause / resume :-)

The only problem is that besides X/Y/Z, E is also locked, so I can't easily unload / load the filament.

I added to the Gcode M18 E, hoping that only E would get disabled, so I can unload / reload the filament, but the reality is that Klipper disables all steppers.

I will probably add the unload on the Gcode instead, to pull the filament out of the hot end, and on the resume, I can add a load and a delay, and that might be a potential workaround.

KevinOConnor, 2018-11-20:
The M18 command (as well as the M84 command) disables all steppers.  It's possible to add a new command (such as DISABLE_EXTRUDER), but it would need to be implemented.

Another option would be to keep retracting until the filament is no longer in the extruder?

-Kevin

cfelicio, 2018-11-20:
Kevin,

Thanks! I just tried tonight, and I was able to do the retraction via the pause (when the printer detects it ran out of filament), but re-loading the filament is another challenge, I'll keep trying to figure out the best way to do it :-)

If not super time consuming to implement, I would love a DISABLE_EXTRUDER command!

Cheers,
Carlos

KevinOConnor, 2018-11-30:
Understood.  It will require someone to implement and submit, however.  If someone is interested in doing that, they should open a github pull request with the code.

-Kevin

cfelicio, 2018-11-30:
I'm interested in doing that, but I haven't touched code for a long time, so I will need to get the rust out first! lol

Would you be able to point me in the right direction, in terms of:

- Is there any similar function that I could use as an example (or base) to get this done? Maybe M18 is a good start, or any other that you can recommend?
- Any files of interest that I will likely need to modify to get this implemented? No need for a comprehensive list, just some pointers that can help me understand Klipper structure a little better, to make sure I put things in the right places

Your help is greatly appreciated!

Cheers,
Carlos

KevinOConnor, 2018-12-09:
There is some documentation about the code layout and high-level code flow at: https://github.com/KevinOConnor/klipper/blob/master/docs/Code_Overview.md

-Kevin

cfelicio, 2018-12-09:
Thanks, will give it a try :-)

closed 2018-12-09 by cfelicio

darrellenns, 2020-01-08:
I would also find this function very helpful.

Looking at the code, it seems like it could be implemented in the PrinterStepperEnable class. The existing M18 handler could be extended to parse the M18 axis parameters (same as marlin does). We would probably also need to add an M17 handler (with axis parameters) to allow re-enabling of steppers.

One thing I'm not clear on is how this interacts with homing. For example, if an axis stepper is disabled and then re-enabled, it should not be allowed to move without re-homing. Or perhaps it should not be allow to be re-enabled by M17 (only by homing). However, the E stepper can be freely disabled and re-enabled. Is the homed/not homed state tracked separately from the stepper enabled state? Or does the code assume a stepper is homed if it is enabled?

The main use case for this would be filament changes/manual extruder moves. A lot of people put knobs on their extruder steppers for this. It's quicker/easier to just turn the knob than it is to fiddle with gcode or lcd menus to make a change. With marlin, you can just "M18 E", turn the extruder knob to load/unload as needed, then "M17 E" and resume printing.

cfelicio, 2020-01-09:
Hi Darrell,

That's exactly the use case at least for me, being able to manually push the filament in place instead of having to use the menu / interface is a time saver. Also, it works quite nicely with the filament runout sensor.

I tried to modify the files to make it work, but I'm not super familiar with Python and could not reach a good solution. Hopefully someone will eventually figure this one out. 

Cheers

ld3300, 2020-02-03:
#2463 Anyone else want to test this out?

cfelicio, 2020-02-03:
Wow for sure, will be more than happy to test! I saw on the commit you found a bug, let me know when it's ready and I'll be happy to try it out! :-)

ld3300, 2020-02-03:
I fixed the issue.  I had to make it so that disabling a stepper didn't make it think the xyz were disabled.  It is working for me now.  There may be a couple of optimizations to future proof it, but I will need some guidance on those.

cfelicio, 2020-02-11:
I just saw it is already commited to master, pretty cool. So to clarify, the final command (in case of a single extruder):

SET_STEPPER_ENABLE STEPPER=extruder ENABLE=0    -> This will disable the extruder
SET_STEPPER_ENABLE STEPPER=extruder ENABLE=1   -> This will enable the extruder 

And it also works for other steppers? Like for X:

SET_STEPPER_ENABLE STEPPER=stepper_x ENABLE=0

ld3300, 2020-02-11:
It will work for any stepper that is setup in the config file.  It does not, however, tell the machine that the stepper is no longer "Homed" if you disable an xyz stepper, so use that part with caution.  `SET_STEPPER_ENABLE STEPPER=stepper_x ENABLE=0` will disable the stepper for x.

It also has ENABLE=1 setup as a default, so you can also just use `SET_STEPPER_ENABLE STEPPER=extruder ` to re-enable the stepper. Although, for any stepper, you don't necessarily have to reenable it.  It will enable as soon as it gets a G1 command, so if you disable the extruder to remove filament you can just leave it disabled until a g-code command tells it to move.

cfelicio, 2020-02-11:
Thanks, I just tested and seems to be working great! Much appreciated! :-)

PuddingBaer91, 2020-03-09:
hi there,

i added this to my config file:

# Idle timeout. An idle timeout is automatically enabled - add an
# explicit idle_timeout config section to change the default settings.
#[idle_timeout]
gcode:

timeout: 10
#   Idle time (in seconds) to wait before running the above G-Code
#   commands. The default is 600 seconds.

but the steppers will be disabled after some idle time. On a delta printer this is fatal. How can i stop klipper disable my steppers on idle?

thank you