Klipper3d/klipper #907: USB Init option?

xC0000005, 2018-11-19:
I have Klipper running on the Malyan M200 V1/MPMini, and this particular board requires a pullup transistor on PB9 to be flipped on before USB init. Would the right way to do this be to add a config to the kconfig and then an #if based off of that?
People wanting to run on this are already going to have to make some changes, because the drivers require a minimum pulse length anyway, and it requires some custom startup assembly to cope with the oddity of the bootloader, so I've got no problems with it just being behind an #ifdef and telling people they need to change it.

xC0000005, 2018-11-20:
> Interesting that you've gotten a 3rd party board with the stm32f103 up and running.
>
> Setting PA12 to low is a hack for the "blue pill" devices that lack a pull up resistor. Using a real resistor attached to a pin (such as PB9) is the correct way to signal a connect. So, I think adding a selection to Kconfig for this is fine.
>
> What delay did you need on the steppers? The default is 2us and that should work on nearly all drivers.
>
> -Kevin

I had marlin running on the mini a year and a half ago, and kept meaning to get around to porting Klipper to it. Once I saw you had an STM32F103 HAL, it was a few minutes of copying over my hard-learned hacks. Value line boards still use the D- pulldown. The Malyan M200V2/Pro, which is an STM32F070 board, for instance, has no pullup, and neither do any of the new L line boards from ST. 

Regarding the delay - 2 should be enough - that’s what I have to set in Marlin, but the behavior is absolutely identical - if I comment out the min pulse in marlin, my steppers buzz just like Klipper does. Maybe I’ve got something configured wrong, but the directions, endstops, heaters, fans, etc, all work. It’s just that the pulses seem to be firing *extremely quickly*. If I just change that (2) value in stepper.c to say, 4, that should be enough for me to test if that’s what’s really causing the problem.

I’ll add a section to the config, and one for the custom startup code (it’s a trivial change to the HAL layer). 

Thanks.

KevinOConnor, 2018-11-30:
Any further update on this?

xC0000005, 2018-12-03:
@clearchris is working on this with the mini. What I saw was that there were to issues probably related to my port - the Z axis moved, but moved erratically when not homing. The whole machine would freeze if I chose to restart the firmware (again, not likely related to the core). One problem I was sure would exist but couldn’t find was that the Malyan boards run at 48mhz and I suspect this was set up for 72mhz.

I’ll leave further updates to him for now.

KevinOConnor, 2018-12-27:
I'm going to close this for now.  I can reopen if there are further questions.  Feel free to create a github pull request with the change as well.

-Kevin

closed 2018-12-27 by KevinOConnor