Klipper3d/klipper #909: stm32f1: First bit of SPI transfer is always zero or a glitch

yschroeder, 2018-11-19:
I am using a blue pill board to configure TMC2130 stepper drivers. While debugging with a logic analyzer I noticed, that the first bit of every SPI transfer (which consists of 5 bytes/40 bits) is always zero.

During configuration this bit need to be high to indicate write access. On my logic analyzer this bits stays zero or glitches to a high signal shortly (but after the clock edge).

I suspect that the SPI circuit in the STM31F1 is not ready yet. As the time between `LL_SPI_Enable(SPI2);` and `LL_SPI_TransmitData8(SPI2, *data);` is too short.

As a workaround I added a `LL_mDelay(1);` after the `LL_SPI_Enable(SPI2);` at the end of `spi_prepare()`. This fixes the problem and the first bit of the SPI transfer is correct. However, it results in all sorts of other artifacts on my logic analyzer (e.g. bogus clock edges outside of SPI transfers).

I am not sure how to fix this correctly, as my knowledge of the HAL interface is rather limited (and I did not find useful information online).

yschroeder, 2018-11-22:
@grigorig: Could you look into this? Or do you need more info? If you could point me somewhere where I can find information about the HAL used, I could look into this myself. For now, I only found a 1800 pages Doxygen that lists all the available HAL functions but does not provide any help on how to call them correctly...

grigorig, 2018-11-23:
I'll take a look at this. FYI I'm only using the low-level (LL) HAL functions. It's a mostly header-only library and basically just a nicer way of reading and writing hardware registers with descriptive names.

KevinOConnor, 2018-11-23:
Thanks.  I wonder if the "*SPI2 = config.config" in spi_prepare() could be causing problems.  (That statement is going to memcpy a bunch of hardware registers and it's unclear if doing so could result in unwanted side effects.)  I think it should be possible to just issue the mode and speed settings directly there.

Separately, I think it would be a good idea to move the spi and adc code from gpio.c to their own files (as is now done on a few other mcus).  Let me know if the work-stm32f1-20181123 branch (specifically commit cb6be270) looks okay.

-Kevin

KevinOConnor, 2018-11-30:
FYI, I went ahead and committed the spi/adc split out from gpio.c (commit 6b108aa8).

-Kevin

grigorig, 2018-11-30:
Thanks. I hope to finally have the time to look at this issue this weekend.

smark-, 2018-12-09:
@yschroeder I am experiencing the same issue and adding delay helps to get the TMC2130 to start driving the steppers. However the motors are very noisy and they start wiggling after few turns. I think it has to do with some synchronization issue due to the added delay. Do you see the same behavior?

yschroeder, 2018-12-09:
@smark- The delay solely makes Klipper being able to configure the stepper drivers at all. After initial configuration, Klipper does not communicate with the drivers again (as far as I know). So any step/direction timing should is not affected by this. Make sure the solder jumpers are configured correctly: https://discuss.toms3d.org/hardware-f6/be-aware-with-chinese-fysetc-tmc2130-t384.html
Otherwise, I believe you have "regular" problems with stepper drivers (current, acceleration).

andrico21, 2018-12-11:
Not sure I have the same issue, but for information: after updating to the v0.6.0-688-gef4f84e (over previous v0.6.0-592-gf6c2e8b) of Klipper I also experience strange issues at least with extruder stepper: looks like the stepper repeately shutting down during the print, skipping steps in the very end of continious extrusion. But I have those issues on Arduino Due, not stm32f1.

closed 2019-01-16 by KevinOConnor