Klipper3d/klipper #910: stm32f1: SPI speed lower than expected

yschroeder, 2018-11-19:
I observed that the SPI speed when communicating with an TMC2130 is lower than expected on the blue pill board. The TMC2130 driver sets the sped to 4 MHz here:
https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/tmc2130.py#L39

On the logic analyzer I observe an SPI clock speed between 2 and 3 MHz. I suspect that the STM32F1 should be able to run much higher SPI speeds, so I think this is a bug. It is not very critical though, as the transfers work nonetheless (apart from #908 and #909).

grigorig, 2018-11-19:
The SPI clock is determined by a simple clock divider, which needs to be a power-of-two. So there isn't much flexibility in clock choice. The base clock of the SPI peripheral is 36 MHz, so in your case, the SPI code should use a divider of 16, which results in 2.250 MHz SPI clock. So I think everything works as designed.

yschroeder, 2018-11-22:
Ok, thank you for the clarification! My feeling was that it should be somewhat more flexible (than on AVR).

Maybe it is worth to point out the available SPI speeds in the documentation for the STM32F1. So nobody gets too confused.

KevinOConnor, 2018-11-23:
FYI, this type of behaviour is pretty common across all the micro-controllers - if they can't reach the exact speed requested, they'll choose the highest speed not greater than the requested speed.

-Kevin

KevinOConnor, 2018-11-30:
If there's no further update on this, I'm inclined to close this issue as there shouldn't be a problem with using a slightly slower SPI speed than the requested one.

-Kevin

yschroeder, 2018-11-30:
Maybe the availble SPI speeds for each platform should be documented somewhere, but it's not that important.

I guess if someone else stumbles upon the problem, he can find this issue.

You can close it I guess.

KevinOConnor, 2018-12-09:
Okay.  It's not particularly easy to document the spi speeds as a number of micro-controllers use "funky formulas" for describing what speeds they can achieve.  It's common to use a speed slightly slower than the maximum value specified in the chip specs - indeed they typically only describe a "max" and don't require any specific frequency likely for that reason.

-Kevin

closed 2018-12-09 by KevinOConnor