Klipper3d/klipper #912: avr: i2c support

trevjonez, 2018-11-20:
the sam4e8e added some level of i2c support for that mcu. I want to add similar support to the avr family of mcu.

I am willing to do the work do you have any guidance on how best to approach the work?

KevinOConnor, 2018-11-20:
I was going to look at this as well.  (I recently obtained an ssd1306 display that I was going to use for testing.)  My rough plan was to port the write-only i2c code currently in src/lpc176x/i2c.c.

-Kevin

trevjonez, 2018-11-20:
Sounds good to me I have plenty to do with my mmu work. I'll check back before I try to dig into this again. I will want to add read support as well so I can hopefully get data flowing back to the host from the prusa filament sensor.

KevinOConnor, 2018-11-24:
FYI - I added write only i2c support (commit 729a2e83).  I don't plan to add read/write support at this time (I don't have a device handy to test read support).  It shouldn't be too difficult to add read support though.

-Kevin

KevinOConnor, 2018-11-30:
I'm going to close this issue as I think the original question was answered.  Feel free to submit a pull request with extended i2c support if needed.

-Kevin

closed 2018-11-30 by KevinOConnor