Klipper3d/klipper #914: missing analog pins on ATMega2560

LowM4N, 2018-11-21:
Hello,

I tried to map the [A2 pin](https://reprap.org/wiki/EinsyRambo_development) of my EinsyRambo as a extruder sensor pin. A thermistor for my inductive probe is connected to this pin, but I get this error message:
```
MCU error during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 140, in _connect
    cb('connect')
  File "/home/pi/klipper/klippy/mcu.py", line 768, in printer_state
    self._connect()
  File "/home/pi/klipper/klippy/mcu.py", line 623, in _connect
    self._check_config()
  File "/home/pi/klipper/klippy/mcu.py", line 574, in _check_config
    config_params = self._send_get_config()
  File "/home/pi/klipper/klippy/mcu.py", line 561, in _send_get_config
    self._name, self._shutdown_msg))
error: MCU 'mcu' error during config: Not a valid ADC pin
```

I found that this shutdown was triggered in this [line](https://github.com/KevinOConnor/klipper/blob/master/src/avr/adc.c#L48). The problem is that the pin A2 is not in [this](https://github.com/KevinOConnor/klipper/blob/master/src/avr/adc.c#L14) static array after the line [28](https://github.com/KevinOConnor/klipper/blob/master/src/avr/adc.c#L28).

I tried to change it to that:
```
static const uint8_t adc_pins[] PROGMEM = {
#if CONFIG_MACH_atmega168 || CONFIG_MACH_atmega328 || CONFIG_MACH_atmega328p
    GPIO('C', 0), GPIO('C', 1), GPIO('C', 2), GPIO('C', 3),
    GPIO('C', 4), GPIO('C', 5), GPIO('E', 0), GPIO('E', 1),
#elif CONFIG_MACH_atmega644p || CONFIG_MACH_atmega1284p
    GPIO('A', 0), GPIO('A', 1), GPIO('A', 2), GPIO('A', 3),
    GPIO('A', 4), GPIO('A', 5), GPIO('A', 6), GPIO('A', 7),
#elif CONFIG_MACH_at90usb1286 || CONFIG_MACH_at90usb646
    GPIO('F', 0), GPIO('F', 1), GPIO('F', 2), GPIO('F', 3),
    GPIO('F', 4), GPIO('F', 5), GPIO('F', 6), GPIO('F', 7),
#elif CONFIG_MACH_atmega32u4
    GPIO('F', 0), GPIO('F', 1), GPIO('F', 2), GPIO('F', 3),
    GPIO('F', 4), GPIO('F', 5), GPIO('F', 6), GPIO('F', 7),
    GPIO('D', 4), GPIO('D', 6), GPIO('D', 7), GPIO('B', 4),
#elif CONFIG_MACH_atmega1280 || CONFIG_MACH_atmega2560
    GPIO('A', 2),
    GPIO('F', 0), GPIO('F', 1), GPIO('F', 2), GPIO('F', 3),
    GPIO('F', 4), GPIO('F', 5), GPIO('F', 6), GPIO('F', 7),
    GPIO('K', 0), GPIO('K', 1), GPIO('K', 2), GPIO('K', 3),
    GPIO('K', 4), GPIO('K', 5), GPIO('K', 6), GPIO('K', 7),
#endif
};
```

It partially worked... I mapped the extruder sensor pin to A2, compiled and flashed the firmware. The weird thing is that I receive the right temp of my inductive probe thermistor, but as the bed temp and the extruder temp was messed up (it shows a random number that was constant).

Than I tried it like that:
```
static const uint8_t adc_pins[] PROGMEM = {
#if CONFIG_MACH_atmega168 || CONFIG_MACH_atmega328 || CONFIG_MACH_atmega328p
    GPIO('C', 0), GPIO('C', 1), GPIO('C', 2), GPIO('C', 3),
    GPIO('C', 4), GPIO('C', 5), GPIO('E', 0), GPIO('E', 1),
#elif CONFIG_MACH_atmega644p || CONFIG_MACH_atmega1284p
    GPIO('A', 0), GPIO('A', 1), GPIO('A', 2), GPIO('A', 3),
    GPIO('A', 4), GPIO('A', 5), GPIO('A', 6), GPIO('A', 7),
#elif CONFIG_MACH_at90usb1286 || CONFIG_MACH_at90usb646
    GPIO('F', 0), GPIO('F', 1), GPIO('F', 2), GPIO('F', 3),
    GPIO('F', 4), GPIO('F', 5), GPIO('F', 6), GPIO('F', 7),
#elif CONFIG_MACH_atmega32u4
    GPIO('F', 0), GPIO('F', 1), GPIO('F', 2), GPIO('F', 3),
    GPIO('F', 4), GPIO('F', 5), GPIO('F', 6), GPIO('F', 7),
    GPIO('D', 4), GPIO('D', 6), GPIO('D', 7), GPIO('B', 4),
#elif CONFIG_MACH_atmega1280 || CONFIG_MACH_atmega2560
    GPIO('F', 0), GPIO('F', 1), GPIO('F', 2), GPIO('F', 3),
    GPIO('F', 4), GPIO('F', 5), GPIO('F', 6), GPIO('F', 7),
    GPIO('K', 0), GPIO('K', 1), GPIO('K', 2), GPIO('K', 3),
    GPIO('K', 4), GPIO('K', 5), GPIO('K', 6), GPIO('K', 7),
    GPIO('A', 2),
#endif
};
```

This time it doesn't show the thermistor(of my probe A2) temp anymore, only the right bed temp was there and the extruder temp was -100 (I think that stays for not connect, so the mapping was broken). I changed only the firmware of the MCU the printer config was still the same. As next i tried to map the right extruder thermistor (in my case F0, that usually works without the change of the firmware) to the extruder sensor pin with my changed firmware where I extended the A2 GPIO, but this stop working too until I remove the GPIO('A', 2). It looks that my change in the PROGMEM break the initialisation of the adc pins, but why? How can i get the A2 pin to get working as a adc pin?

The mapping of the A2 pin to the extruder was only a test for thermistor. I want write later a module which waits with the next gcode command until a desired temperature reached at my inductive probe.

KevinOConnor, 2018-11-23:
In order for us to assist, you'll need to attach the Klipper log file from the event - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

If you've modified the code, please revert to an unmodified version of Klipper when producing the above log.

-Kevin

LowM4N, 2018-11-23:
Hello @KevinOConnor,
I think you missunderstand me. I don't have a problem with the current release of Klipper, it runs well on my custom Prusa MK3. On my board the EinsyRambo is a AD pin (the PA2), a thermistor is connected there. I need the value of this thermistor to optimize my z-offset when probing.

My question is how can I archive this in a Klipper Module?

KevinOConnor, 2018-11-26:
The atmega2560 hardware is not capable of making an ADC measurement on the PA2 pin.  The log makes it easier for us to answer your questions - please do provide it for any future inquiries.

-Kevin

LowM4N, 2018-11-26:
@KevinOConnor You are right, the PA2 Pin is really not a acd pin. I had wrong information out of [this ](https://reprap.org/wiki/EinsyRambo_development) wiki. In the schematic [here](https://github.com/ultimachine/Einsy-Rambo/blob/0.5a/board/Project%20Outputs%20for%20Einsy_Rambo_0.5c/Schematic%20Prints.PDF) is it PF3! And it works now as expected.

closed 2018-11-26 by LowM4N