Klipper3d/klipper #916: mmu2: stepper pin layout discussion.

trevjonez, 2018-11-21:
@KevinOConnor my goal with this issue is just to have a place to discuss the right way to implement the work for getting the mmu2 working cleanly with klipper.

So on the prusa mmu2 there is three tmc2130 drivers. all of which receiver their direction and enable signal via pins from a 16-bit configuration of 74HC595 units.

the step pin and spi are as you would expect directly off the 32u4.

as it is it seems that the stepper.c class is setup to control the step/dir interface directly via GPIO pin refs from the gpio_out struct. If I follow that setup flow it looks like the `gpio_out_setup` sets up the struct based on the compile time register and pin layout. From there perhaps I could extend that logic to check if the pin def is a virtual pin before failing out?

so my intuition is telling me that it should be possible to create a way to declare a set of virtual pins in the printer.cfg that would look something like this:
```
[74hc595 sr1]
depth: 16
data_pin: pb5 # ic pad 14
latch_pin: pb6 # ic pad 12
clock_pin: pc7 # ic pad 11
out_enable_pin: p** # ic pad 13. optional param?
clear_pin: p** # ic pad 10. optional param?
```

which would result in virtual pins `sr1_0`, `sr1_1`, `[...]`, `sr1_F` that could then be used to define the other steppers in a way that is consistent with the current mechanism?

```
[stepper_s]
step_pin: pd4
dir_pin: !sr1_2
enable_pin: !sr1_3
[...]
```

I would then assume that to make this work I would need to allocate 1 bit per virtual pin to act as a pseudo register then the implementation of the struct provided by the gpio setup would push those bits out. which means that most/all of the gpio_out functions would need to be rewritten to know how to deal with the virtual pins via shift register?

and then finally does the overhead of sending data out to the shift register mean that the kinematic system would need to be aware of it and adjust timing to compensate for the slower output signaling? in this case a virtual dir pin means that a direction change would have what 6-7ms of overhead?

how far off am I? This type of stuff is so far out of my comfort zone.

KevinOConnor, 2018-11-22:
> From there perhaps I could extend that logic to check if the pin def is a virtual pin before failing out?

It's possible to do that, but I'd recommend against it.  I fear it would add overhead for normal users.  

Another option would be to manually perform direction changes in the host code (by explicitly sending the shift register update on any direction change).  Assuming the mmu2 doesn't change directions often, this shouldn't be too bad.

> so my intuition is telling me that it should be possible to create a way to declare a set of virtual pins in the printer.cfg that would look something like this:

Klipper has a standard mechanism for defining pins on separate chips - it's with a chip: prefix.  For example, the replicape uses `replicape:stepper_y_enable` type definitions.  So, I expect the mmu2 python module could define pins with something like `mmu_shift_register:stepper1_enable`.  Take a look at how klippy/extras/replicape.py calls ppins.register_chip().

As above, this system would prevent one from defining the direction and step pins on different chips.  But, you should be able to use a dummy pin (eg, PF3) for the direction pins of your steppers for now.

> and then finally does the overhead of sending data out to the shift register mean that the kinematic system would need to be aware of it and adjust timing to compensate for the slower output signaling?

I'm not familiar with how the steppers in the mmu2 are utilized.  It's not necessary to register the steppers with the kinematic system though - one can just manually control them.  Take a look at the klippy/extras/force_move.py:manual_move() code for an example of this.

-Kevin

KevinOConnor, 2018-12-17:
Any further comments on this?  If not, I'm inclined to close this issue.  (We can always reopen it if there are further questions.)

-Kevin