Klipper3d/klipper #917: Support LED strip control with M150

gmag11, 2018-11-21:
Hi,

I have an RGB LED strip (not Neopixels) connected to PWM pins on my board. I use M150 command to control it.
Could it be possible to add support for this on Klipper.

Congrats for such a great development!

KevinOConnor, 2018-11-23:
It should be possible to control the pwm pins using the "[output_pin]" config and the SET_PIN g-code command.  See the config/example-extras.cfg file for details.

If you wish to use M150 instead of SET_PIN, then it is possible to create a [gcode_macro M150] config to run SET_PIN on an M150 request.  Again, see the config/example-extras.cfg file for the details.

-Kevin

KevinOConnor, 2018-11-30:
I'm going to close this issue as I think the original question was answered (use a gcode_macro).

-Kevin

closed 2018-11-30 by KevinOConnor

opit7, 2018-12-28:
Hello,

is there a way to control Neopixels/Ws2812?

-opit

Zwaubel, 2019-05-28:
Hi @KevinOConnor,

could you elaborate on what it takes to implement support for Neopixels/WS28xx LEDs to be used as event driven pixels, for instance as an indicator for temperatures (nozzle and/or bed), print progress etc.?

Thanks!

KevinOConnor, 2019-05-28:
The Neopixel (and similar) would require host python code to be written.  (The original question was about pwm controlled leds.)  If one wishes to automatically update the led (neopixel or other) whenever some printer status changs then host python code would also need to be written to do that.

-Kevin

theopensourcerer, 2019-05-28:
I'm fairly sure you can do something like this using the Octoprint Enclosure plugin:

https://plugins.octoprint.org/plugins/enclosure/

"Use custom Gcode to control neopixel"

rykiop, 2019-05-29:
I would much rather use the main control board to handle everything "printer" related. Having ws2812 LED integration for mcus for Klipper would be the bee's knees. It takes chopping up the timers or halting them all together for a few millis.

KevinOConnor, 2019-07-18:
FYI, Klipper now has some basic support for Neopixels.

-Kevin