Klipper3d/klipper #920: Error: Internal error on command:"PID_CALIBRATE

vsusanto, 2018-11-22:
I am using Octoprint with Klipper on Raspberry Pi 3.
I received this error during PID adjustment process:

Recv: // Klipper state: Shutdown
Recv: !! Internal error on command:"PID_CALIBRATE"
Changing monitoring state from "Operational" to "Error: Internal error on command:"PID_CALIBRATE""
Changing monitoring state from "Error: Internal error on command:"PID_CALIBRATE"" to "Offline (Error: Internal error on command:"PID_CALIBRATE")"
Connection closed, closing down monitor

I appreciate the help.

theopensourcerer, 2018-11-22:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

vsusanto, 2018-11-22:
Sorry forgot the log. Here it is.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2608881/klippy.log)

vsusanto, 2018-11-22:
I found the solution, but it might not be an elegant way to solve it, but so far it works.
I inserted these lines into the configuration file:
[verify_heater extruder]
heating_gain: 10
check_gain_time: 30
hysteresis: 10
max_error: 120

I suspect the extruder is very unstable that cause this issue. So far, I was able to perform PID tuning and save the configuration.

closed 2018-11-22 by vsusanto