Klipper3d/klipper #922: MKS sbase multiple steppers

nandinho84, 2018-11-23:
I'm trying to add 2 more steppers (7 in total) with two external stepper motor drivers. I'm trying to use the J7 (P0.14, P0.16, P0.17)and J8 pins (P.1.22, P1.23, P2.12) on MkS without success. It seems that those pins aren't working, or at least some of them aren't, because the stepper doesn't move at all.

KevinOConnor, 2018-11-25:
You'll need to attach the klipper log file in order for us to assist - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

nandinho84, 2018-11-25:
OK, sorry. Turns out, that mks sbase has some pins that can be wrong on the pinout table... I've chosen different pins (more common) and the problem was fixed.

closed 2018-11-25 by nandinho84