Klipper3d/klipper #924: Klipper stopped midprint with "invalid oid type"

benhurstein, 2018-11-23:
Hi!
I use klipper for some months now, and this is the first time I have seen this error:
"MCU 'mcu' shutdown: Invalid oid type"

The only difference from last print was that before starting this print job I changed the extruder sensor type:

#sensor_type: Honeywell 100K 135-104LAG-J01
sensor_type: NTC 100K beta 3950

[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/2611822/klippy.log.gz)

KevinOConnor, 2018-11-24:
The log indicates the micro-controller went through an unexpected restart just prior to the shutdown - that restart is the root cause of the error above.  It's unclear what the cause of that restart is, but a hardware fault is a likely explanation.

The log also indicates that there are sporadic message retransmissions occurring throughout the print (the bytes_retransmit counter periodically increases).  This is quite unusual on a USB connection.  Extreme electrical noise could be an explanation.  I recommend you double check your wiring.

It's possible the issue could be due to voltage fluctuations.  If so, reducing the max_power in your heater_bed config may help (eg, by setting max_power=0.5).

-Kevin

benhurstein, 2018-11-24:
Thank you.
Maybe my raspberry is overheating, it has been a very hot day here yesterday.
I'll increase it's ventilation (and double check the wiring) and try again.

Benhur

closed 2018-11-24 by benhurstein