Klipper3d/klipper #925: [Feature request] z probe connected to Raspberry Pi

varazir, 2018-11-24:
Hello,

Don't know if this has been asked before or if there are support for it already.
I have a printer that don't support z-probe on the MCU so I have an idea/question can you connect the probe to Raspberry Pi?

KevinOConnor, 2018-11-25:
Some development work would be needed to make this possible (the klipper linux mcu code would need to support the raspberry pi gpios).  So, it's not currently supported.

-Kevin

varazir, 2018-11-25:
> Some development work would be needed to make this possible (the klipper linux mcu code would need to support the raspberry pi gpios). So, it's not currently supported.
> 
> -Kevin

Ah okay, something that can happen ?

CageFox, 2018-11-25:
40 unused gpios - i also think it can be very interesting ability. If extrapolate possibilities a little we can see this way can lead us on "3d printer solely on raspberry"

varazir, 2018-11-26:
Changed the subject to reflect more to what we want rather just z probing

derbengale, 2018-11-26:
+1 for the GPIO support 👍

supermerill, 2018-11-26:
Problem is, rasbian isn't a real-time os. So if you want to probe with the raspberry, it may be less precise than on a real-time mcu. I can only see these gpios used for task that are not time-critical, like temperature  sensing or fan control.

CageFox, 2018-11-26:
This makes a sense. And if printer makes home movement and raspbian delays for second for something - the bad, really bad things may happen

But ability to control gpio may be useful for some less critical tasks. Fans, main psu turn on/off, robotic camera arm movements, etc

jakep82, 2018-11-26:
All of these things can already be accomplished using an existing Octoprint plugin.

https://plugins.octoprint.org/plugins/enclosure/

CageFox, 2018-11-26:
Octoprint is Octoprint and Klipper is Klipper. What if i using klipper with repeiter server, because Octoprint can not take heavy gcode load.

jakep82, 2018-11-26:
Octoprint is the supported way to run Klipper.  I'm sure if you or someone else wants to write code to support this it could be merged, but it seems like a lot of effort when someone else has already done what you want.  Also, the issues you mention with gcode in Octoprint are fixed by using the virtual SD feature.  I've successfully used Klipper on Pi Zero W running Octoprint using the virtual SD feature.

CageFox, 2018-11-26:
Using virtual sd excludes octoprint from process, there is no web control, no time estimations, etc. I tried it and were not very happy. It's like a quick fix for something not working with duct tape.

jakep82, 2018-11-26:
I'm not exactly sure what you mean.  I use virtual SD exclusively.  I do have full control of my printer using the web interface, and I do get time estimations.  The only difference I notice is it doesn't display the current Z height.

CageFox, 2018-11-26:
>  I use virtual SD exclusively. I do have full control of my printer using the web interface, and I do get time estimations. The only difference I notice is it doesn't display the current Z height

Strange, when i started print from virtual sd card i received no information about print from octoprint - only "printing now". No Z pozition, no estimated time (corrected while printing process go), no graphics in "GCode viewer", only temperatures.

How octoprint can receive this information after all? Printing from sd is when klipper print all g code on his own, klipper not reporting back to octoprint what it's doing and where he is now

CageFox, 2018-11-26:
I ever think when printing from virtual sd octoprint can be completely shutted down and print will continue without problems

@KevinOConnor Can it be done?

CageFox, 2018-11-26:
In any way it's not a point. Klipper is rather friendly and can work not only with octoprint. I personally not very happy with octoprint and plan to test klipper with repetier server

KevinOConnor, 2018-12-09:
I'm going to close this issue as I believe the original question was answered (Klipper does not currently support toggling gpio pins on the raspberry pi).  I personally don't have a short-term plan to add support for this and it doesn't appear anyone else is actively working on this.

Also, note that controlling stepper motor drivers from an rpi would be dramatically more work than just adding support for toggling gpios for general purposes (such as to blink leds or control heaters).  Since a probe needs to be on the same "micro-controller" of the stepper motors that it halts during the probe process, it is highly unlikely there will be rpi support for probes.

I will reopen if someone starts active development on rpi gpio support.
-Kevin

closed 2018-12-09 by KevinOConnor

varazir, 2018-12-09:
I'm ok with closing this. 

Set it in a wish list if you have any :)

KevinOConnor, 2020-06-01:
FYI, this support was merged as of #2787 .

-Kevin